bambulab/BambuStudio #1433: [feature Request] Have X1C camera working in LAN Only mode

BambuLabDiscord, 2023-03-08:
This request is simpler than other requests related to the camera. The request is to have the camera working in LAN Only mode so users can take quick visual checks without having to stop what they are doing and leave the desk.  It's fine if the camera only works on BS desktop version. This request also doesn't ask to have video streams available outside of LAN.

Having the camera working is  helpful because even in LAN mode, the printer can be in another room, garage, or the same room but not close enough to see how the print is going. It also reduces the need for users to go near the printer while printing volatile filaments that emit unpleasant fumes/particles.

lanewei120, 2023-03-14:
the same as https://github.com/bambulab/BambuStudio/issues/1372

AkitaAttribute, 2023-04-26:
Bumping this cause you've linked a dead issue.  
[image]

AkitaAttribute, 2023-04-26:
@lanewei120

lanewei120, 2023-04-27:
it seems the owner deleted that issue

sodachen22, 2023-04-27:
https://github.com/bambulab/BambuStudio/issues/1536

LayerCakeMakes, 2023-04-27:
> Bumping this cause you've linked a dead issue.

There was some drama in that issue where the author tried to change the issue afterwards to include more and different issues. People tried to get it back on track until the author deleted it and seemingly left github. The new issue is #1536 as mentioned above just giving some context.

AkitaAttribute, 2023-04-27:
👍 Didn't see that originally, and since we had the outage yesterday, I wanted to make sure it was still on the radar.

kriordan37, 2023-06-05:
need this

chrisipa, 2023-09-20:
This is desperately needed ...

SebiPanther, 2023-09-23:
Where should i start? Its Bullshit to block the Camera in Lan mode. That just a inconvenience to keep the cloud stuff enabled and risk your printer that way: https://all3dp.com/4/bambu-lab-cloud-error-causes-prints-to-start-unprompted-users-printers-damaged/
The last move makes me especially angry. I can't use the App anymore until i join there try of of the next shitty social network surrounding 3D Prints and making model inaccessible without a account. And thats just the tip of the iceberg what is wrong with that copy cat 3d cults platform i was never interested in using anyway. That is not even original. But it also renders the Camera useless on my phone. I can't even stop the printer anymore if a prints fails. What it does often due to the quality of the product.
That habit of companies forcing me to deal with there try to get even more out of me, after paying over thousand dollar for a shitty working printer in the first place should be illegal.
And all my tries to get rid of that shitty machine have been unsuccessful due there "policy" (aka. keeping money and pump more garbage into the world).
This gets probably deleted like the request of getting camera support in lan mode anyway. Its the lack of decency with that company that really makes me angry!

AkitaAttribute, 2023-09-23:
> Its the lack of decency with that company that really makes me angry!

You gotta chill man.  Just give them a garbage gmail address.  We all have spam email addresses.  Don't drag the topic off into la la land.  We want LAN camera functionality, not an ignored topic because you're angry.

SebiPanther, 2023-09-23:
> > Its the lack of decency with that company that really makes me angry!
> 
> You gotta chill man. Just give them a garbage gmail address. We all have spam email addresses. Don't drag the topic off into la la land. We want LAN camera functionality, not an ignored topic because you're angry.

The garbage Mail account does not work if your models (and those of others) get stored in there cloud. But i agree with you - this threat is about the camera in Lan Mode. Reason why i explained in detail two cases where it is super bad to not have the camera. Specially without there copy cat Social Network.
But yeah - if that "feature" ( to even call it that way feels wrong - it takes afford to disable that feature in the first place ) i will have a toast to Bambulab or something.

SebiPanther, 2023-09-24:
Turns out: at least on the X1C is a work around: https://www.wolfwithsword.com/bambulabs-x1c-camera-in-home-assistant/
But better to just start here: https://github.com/greghesp/ha-bambulab

WZ9V, 2023-11-20:
Have same issue with X1C camera in Linux Mint 21.2 Cinnamon in LAN Mode.  Plays for a few seconds and then locks up.