bambulab/BambuStudio #1434: G-Code for Filament Unload/Load

EukaDesigns, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
It would be useful to know the filament unload/load g-code. This would allow implementing filament changes for printers not using an AMS. 

**Describe the solution you'd like**
A copy of the filament unload and filament load g-code. I'm uncertain where the load g-code would be, but I believe the unload g-code is located here: "/usr/etc/print/filament_unload.gcode"

**Describe alternatives you've considered**
I have tried to mimic the motions used and recreate the g-code on my own for the filament cut and unload procedure, but I would feel more comfortable using the g-code the printer already uses.

Gr4sil, 2023-03-09:
Hello,

for the AMS the unload and the load gcodes can be found here in github:
https://github.com/bambulab/BambuStudio/tree/master/resources/printers

I have also thought to tinker new printer profile with filament change codes to work without AMS.

EukaDesigns, 2023-03-09:
@Gr4sil
I was aware of the AMS load/unload g-code files, the issue is that these call T[next_extruder], which does not work without an AMS. The g-code specifically for filament cutting for unload, and filament sensing for load are not explicitly stated. For the load and unload g-code that is called when you press the "unload" and "load" buttons on the machine itself, I would assume that all of the steps would be explicitly stated. 

For the unload g-code, I have attempted to recreate the motions taken for cutting the filament, and for extruding forward a bit before rewinding the filament out; however, I would feel better about the code if I knew what the values BambuLab determined for the speeds and distances. If you're curious to see it, I've posted it here: [change_filament_noAMS.gcode](https://github.com/EukaDesigns/Bambu_CustomGCode/blob/bfdd8d4e9dce94def22d21b0e4ac8b5e49ebadeb/change_filament_noAMS.gcode)

For the load g-code, I have no idea how to wait for the filament detection before continuing. I assume it would be some variation of M400, but not sure what exactly.

StoneLiBambulab, 2023-03-10:
Hi @EukaDesigns ,
in the BambuStudio installation path, for example the "C:\Program Files\Bambu Studio\resources\printers" 
there are ams_load.gcode and ams_unload.gcode

Thanks!

Stone Li

EukaDesigns, 2023-03-12:
Hi @StoneLiBambulab!

I did see the AMS unload and load gcodes, but I was looking for unload and load g-codes that are called when you press the button to do it in either Bambu Studio or on the P1P directly. I'm guessing this code is stored on the printer in the firmware, so it might need to come from the firmware team.

StoneLiBambulab, 2023-03-13:
Hi @EukaDesigns ,
ok. you are right. i don't have it now, and i can try to get it from printer team. I will leave a message for any reply to me.

Thanks!

Stone Li

EukaDesigns, 2023-03-17:
Hi @StoneLiBambulab,

The g-code provided in the zip file is an exact copy of the ams_unload g-code, and is not the same "filament_unload.gcode". I'm sorry to keep asking, but could you clarify to the printer team that you're looking for the g-code for the manual unload/load, and not the AMS unload?

StoneLiBambulab, 2023-04-03:
hi @EukaDesigns ,
   Maybe you want an old version, the new version has been modified to this.

Thanks!
Stone Li

Gr4sil, 2023-05-13:
hi @StoneLiBambulab 

Is it in BambuStudio roadmap to let users make the filament changes without AMS? That is the target of @EukaDesigns approach, so that we could make filament changes without AMS. And yet better would be possibility also co-operate with AMS, so that users with AMS and Y-tube could when necessary to use for example TPU from external spool.

StoneLiBambulab, 2023-05-28:
[image]

you can add a pause and change the filament manually at specified layer.

we will condider supporting to switch the filament between ams and external spool . But the printing process requires pulling out the filament from external spool manually, which is not friendly.

Thanks!

Stone Li

Marcel40625, 2023-06-25:
Yeah give us the possibility to change filament with "Filament end g code" or something for external spool swap please! printing with 2 colors TPU is a hastle ...

adripo, 2023-07-20:
> Hi @EukaDesigns , here is the unload gcode [unload.zip](https://github.com/bambulab/BambuStudio/files/10968573/unload.zip) and no load gcodes.
> 
> Thanks!
> 
> Stone Li

Hi @StoneLiBambulab I checked the code you posted and noticed that the movement to cut the filament is missing. Can you please provide the gcode for filament cut movement? Thank you!

orestis, 2024-10-29:
I would be really interested in this as well. Some simple models might only need a couple of filament changes that are definitely possible without an AMS.

MrMonkey1302, 2026-03-25:
@StoneLiBambulab did you ever manage to find the movement gcode for cutting the filament?