bambulab/BambuStudio #1435: mac using bambu, software crashes

eagle-ice, 2023-03-09:
The problem occurs when setting the wall to 0。

Detailed information and system configuration 
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [3559]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.04.02.13 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-03-09 14:48:09.7430 +0800
OS Version:            macOS 13.2.1 (22D68)
Report Version:        12
Anonymous UUID:        28C6FE47-C220-6FE5-AC77-A9740F651010

Time Awake Since Boot: 8800 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000b0990000b099 -> 0x000030990000b099 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x0000b0990000b099

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [3559]

VM Region Info: 0x30990000b099 is not in any region.  Bytes after previous region: 52952651837594  Bytes before following region: 52119428091751
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Kernel Triage:
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage
VM - pmap_enter retried due to resource shortage

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   BambuStudio                   	       0x100b01e4c Slic3r::GUI::GCodeViewer::update_marker_curr_move() + 468
1   BambuStudio                   	       0x100b01e28 Slic3r::GUI::GCodeViewer::update_marker_curr_move() + 432
2   BambuStudio                   	       0x100a11f6c Slic3r::GUI::GLCanvas3D::_render_gcode(int, int) + 488
3   BambuStudio                   	       0x100a0f8ac Slic3r::GUI::GLCanvas3D::render(bool) + 1296
4   BambuStudio                   	       0x100a18888 Slic3r::GUI::GLCanvas3D::on_idle(wxIdleEvent&) + 408
5   BambuStudio                   	       0x10294ceb8 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
6   BambuStudio                   	       0x10294cb84 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
7   BambuStudio                   	       0x10294c9ac wxEvtHandler::ProcessEvent(wxEvent&) + 56
8   BambuStudio                   	       0x10294d064 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
9   BambuStudio                   	       0x102768468 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 92
10  BambuStudio                   	       0x102768490 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
11  BambuStudio                   	       0x102768490 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
12  BambuStudio                   	       0x102768490 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
13  BambuStudio                   	       0x102768490 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
14  BambuStudio                   	       0x1026a0eec wxAppBase::ProcessIdle() + 148
15  BambuStudio                   	       0x10283eaa8 wxApp::ProcessIdle() + 36
16  BambuStudio                   	       0x102955c24 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 88
17  CoreFoundation                	       0x1993dcdb8 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
18  CoreFoundation                	       0x1993dcca4 __CFRunLoopDoObservers + 532
19  CoreFoundation                	       0x1993dc3e4 __CFRunLoopRun + 1048
20  CoreFoundation                	       0x1993db878 CFRunLoopRunSpecific + 612
21  HIToolbox                     	       0x1a2abbfa0 RunCurrentEventLoopInMode + 292
22  HIToolbox                     	       0x1a2abbde4 ReceiveNextEventCommon + 672
[... 471 lines cut ...]

      },
      "deploymentId" : 240000021
    }
  ],
  "experiments" : [

  ]
}
}

Model: Macmini9,1, BootROM 8419.80.7, proc 8:4:4 processors, 16 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: DELL U2722DX, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Nov 30 2022 03:35:43 version 18.20.379.9.7.8.145 FWID 01-9578a936
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB30Bus
USB Device: 2.4G Device
USB Device: wireless dongle
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

SlawaGurevich, 2023-03-09:
Same thing here. However, for me it just crashes after I press "Print Plate" if my printer is in LAN only mode.

StoneLiBambulab, 2023-04-03:
Hi @eagle-ice ,
    it is crash in Gcode Viewer. The problem will be easy to solve if there is a reproduce method. I tried to reproduce the crash, but crash did not happen. Could you help upload your 3mf file for me?

Thanks!
Stone Li

StoneLiBambulab, 2023-04-06:
Hi，
    i will add some code to avoid crash.

Thanks!
Stone Li

Hologos, 2023-04-15:
Hello, it has happened to me several times as well. Here are the crash log + the 3mf file.
[bambu-studio-crash.zip](https://github.com/bambulab/BambuStudio/files/11239211/bambu-studio-crash.zip)

eagle-ice, 2023-05-28:
> 
Sorry, 3mf can't be found.