bambulab/BambuStudio #1439: Allow bambustudio to support older MacOS X versions

Babsobar, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
The problem is that I can run prusaslicer on my older Mac, that is my workshop computer, but cannot run bambustudio on it.

**Describe the solution you'd like**
I'd like bambustudio to be able to be installed on Mac OSx sierra 10.12.5

**Describe alternatives you've considered**
Using another more recent computer is impractical because the computer and the printer are not in the same place

neilhigbee, 2023-04-27:
I hope they do compile for older versions 
As would love it for El-Capitan
As I have a mac pro 2.1 in my workshop

lanewei120, 2023-04-28:
hello,

we are sorry we don't support macos 10.12,
as we can not find a MacOS 10.12( we tried to find some macos with old version before and we could not get a 10.12)
it is hard for us to do it without verify

And previously we found some libaries(not used by prusa) requiring at least 10.15,
so we set the minimal version of MacOS to 10.15

really apologize for this

Babsobar, 2023-04-29:
Thank you for your response,

Seeing as I thought it was not going to be easy for you, I managed to find
a workaround and to turn my old 2009 mac into a 11.X machine using
hackintosh software.
I can now use Bambustudio on it, even though performance is quite poor