bambulab/BambuStudio #1441: Enhancement : Make view options more accessible

ipodsync, 2023-03-09:
Presently, the view options like top, left, perspective, orthographic, etc... are buried 3 levels down in the menu. I use these commands frequently and wish they were more accessible.

Showing the view menu in the menu bar would be very helpful.   Alternatively, having these commands available in the main toolbar would also work.  Even in the right-click menu in the work area would be good.

ipodsync, 2023-03-14:
Shortcut keys are handy ... making them more accessible on the UI is also handy.

I run on both Mac and Windows and the Mac version is much more accessible because of the Mac's menu bar.  

The title bar on windows is mostly empty.  It would be nice if the view menu was just there like the file menu is.

lanewei120, 2023-03-14:
thanks for the suggestion.
we will discuss it internally

julie777, 2023-04-01:
@ipodsync 
The menu system is non-standard and confusing in general. I submitted #1561 for this.