bambulab/BambuStudio #1442: BambuStudio adds holes in the model.

Babsobar, 2023-03-09:
**Describe the bug**
In prints that have thin walls, Bambustudio adds holes in the design. Turning "detect thin walls" on adds holes in other places.

**3mf File for This Bug**
Github does not allow the sharing of .3MF files. I will share the whole of the project in a dropbox or whatever.
https://file.io/cGhxBk59vLJh

**To Reproduce**
In fusion 360, Make a file in which the infill is part of the design of the piece, meaning that you have manually designed the infill as part of the piece needed to be printed. Shell the piece so that the outside walls are 0.45 mm. 
The use case for this is for 3d printed parts that are transparent. See photos. 
Set filament to PLA, and 0,16mm OPTIMAL profile. Set infill to 0%. Slice. Send to printer over the internet, using a new roll of natural PLA from fillamentum.

**Expected behavior**
Have bambustudio keep the file as it is meant to be! This does not happen in prusaSlicer. Also, the print quality of the surface is absolutely terrible! It produces "scratches" on the surface that are protruding almost 0.5 mm. See photo. 

**Screenshots**
[image]
[image]
When detect thin walls is on:
[image]
[image]
[image]
Terrible print surface quality 
[image]
[image]

**Desktop (please complete the following information):**
 - Windows 10, latest updates
 - Latest Bambulab version

I bought this machine as a professional artist, because I wanted to have a more detailed, faster, and modern machine, but compared to when I had a Prusa MK2S and I was using Prusaslicer, I have been having more problems. I know that what I print is not the usual but this machine is supposed to be a professional tool and right now, it really is putting a strain on my professional activity because of strange problems like this that I didn't have before.

SaltWei, 2023-03-14:
> Github does not allow the sharing of .3MF files. I will share the whole of the project in a dropbox or whatever.
> https://file.io/cGhxBk59vLJh

Can not get the file. You can just rename the 3MF as .zip

Babsobar, 2023-03-14:
Thank you for your answer, here is the file zipped up.

[Troubleshooting.zip](https://github.com/bambulab/BambuStudio/files/10972201/Troubleshooting.zip)

SaltWei, 2023-03-15:
[image]
[image]
[image]
It is just caused by too thin model. Not slicer bug.
Changing settings as pictures can help.

Babsobar, 2023-03-16:
[image]
This is the same file in Prusaslicer, as you can see, no holes are generated. 

[image]
I have tried your setting in Bambustudio, the only one that helps is Arachne, but I need the printer to slow down for overhangs because I can't use supports for this print. Seeing as this is a transparent print and has 

"It is just caused by too thin model."
I have printed files and sliced models with these settings since 2016 in prusaslicer, so it's not that the model is too thin. I have had success with a MK2S and prusaslicer. As you can see in the pictures.

SaltWei, 2023-03-16:
> but I need the printer to slow down for overhangs because I can't use supports for this print.

I have printed your 3mf and only changed the settings in my screenshot, and printed successfully.

SaltWei, 2023-03-16:
OK. The original 3MF has been slowing down already

Babsobar, 2023-03-16:
> > but I need the printer to slow down for overhangs because I can't use supports for this print.
> 
> I have printed your 3mf and only changed the settings in my screenshot, and printed successfully.

Did your print show the same surface quality? I have no clue what that is due to

SaltWei, 2023-03-16:
Return to the tickets. The hole is just caused by different settings of slicer.
If you use same classic perimeter generator and disable thin wall, prusa also has problem.

Babsobar, 2023-03-18:
Hello, 
I have fixed the problem. 
Turns out, none of the setttings were the problem for the holes appearing through the print, after a long investigation and trying out a lot of different settings in prusaslicer and BS Lab, I have figured out that the culprit was the mesh exporting algorithm in Fusion360; even though I had the setting on "Refined". The algorithm was making bad faces at the spot were the holes were appearing. While Arachne was able to fix the holes, the resulting prints were not very good in surface quality compared to classic slicing.

I am printing a new print as we speak and will keep you updated on the surface quality.

Thank you for your time and patience.

closed 2023-04-13 by SaltWei