bambulab/BambuStudio #1443: [*FEATURE REQUEST*] Bambu Studio API

BambuLabDiscord, 2023-03-09:
This is a big ask.  The CLI for PS/Slic3r leaves some to be desired.  But if there were a way I could programatically interact with most of the features of Bambu Studio -- including the text tools, I believe this would solve a serious gap in the market for cottage industry 3d printing businesses (small to medium print farmers - up to the bottom of industrial 3d printing).

programbo, 2023-07-01:
I would also love an API, even if it's just for monitoring and maybe Pause/Resume/Cancel. Ideally there should already be a robust, authenticated, API that Bambu Studio consumes.

Happy for Bambu Lab to focus their effort on the core products and let third-parties to create an ecosystem of integrations.

VictorDxx, 2023-11-25:
would like this feature as well!

OrhanYigitDurmaz, 2024-09-20:
ping?