bambulab/BambuStudio #1444: AMS filament backup not working on P1P

Shareisfun, 2023-03-09:
P1P OTA version01.02.01.00（20230302）

Has anybody got AMS filament backup working?

I havent, and i have to log the printer out of the slicer and back in to enable it but i isent working and i’m using beta slicer? When i close the slicer i have to log the printer out and back again to be able to choose it, but it not working`

[Picture of AMS](https://scontent-cph2-1.xx.fbcdn.net/v/t39.30808-6/334712680_150109854592214_7354183288760062298_n.jpg?_nc_cat=103&ccb=1-7&_nc_sid=5cd70e&_nc_ohc=SH99wYAL-cYAX_00lIF&_nc_ht=scontent-cph2-1.xx&oh=00_AfBPhyp8Ocx7dZEgUOSEeX0hlEVscgrB1vri3Mn4OG4Zkg&oe=640E269C)

edit: Now i’m on version 1.5.0.60 as off to day but still not working.

ab3dgeek, 2023-03-10:
it worked for me just a few hours ago.

bambu studio 1.5.0.60
p1p fw: 01.02.02.00
ams fw: 00.00.05.96

Shareisfun, 2023-03-10:
I have the same setup and it is not working for me?
[Video](https://drive.google.com/file/d/16UveoT8CL7HrIEkN54nrcdrZkqbWyo7z/view?usp=sharing)

walterwongbbl, 2023-03-15:
[image]
[image]

Hi, Because you are currently selecting a virtual tray.  
you can select AMS tray and try again

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl