bambulab/BambuStudio #1445: [*FEATURE REQUEST*] Import Images to the paint tool

BambuLabDiscord, 2023-03-09:
As a user, I'd like the ability to import various image types (jpg/png, svg) to use as a *mask* and/or to use directly to painting (match closest colors in AMS would be amazing, considering you can daisy chain those suckers). ;)

lanewei120, 2023-03-14:
currently svg file is already supported in Bambu Studio
[image]

MadMax389, 2023-03-14:
I think what the OP is requesting is a way to "stamp" the image onto the model. Mosaic's Canvas has this feature.
1. Import the .svg and assign colors to the image.

[image]
[image]

2. Place the image on the model. Rotating and scaling available.

[image]

3. Now the image has been "painted" on the model ready for slicing.

[image]

AiolosIvor, 2024-06-29:
Oh man, this is exactly what I'm looking for!

GCRVL, 2024-08-13:
Essential feature to make the paint tool useful.

eried, 2024-12-02:
Still waiting for this 😔