bambulab/BambuStudio #1446: Reduce print start time by reordering heat control gcode

julie777, 2023-03-09:
# Is your feature request related to a problem? Please describe.

BambuLab printers advertise very fast printing speeds a major feature. However, the startup time before print starts is extremely long compared to other printers. In fact, for a small print the startup time can be longer than the print time.

# Describe the solution you'd like
Currently the second step of the start gcode is to set the temp for the bed and then wait until temp is reached. This wait is just unnecessary. I realize that at some point the code has to wait until the bed temp is correct, but not here.

Other printers
- set the bed temp
- set the nozzle temp
- wait for both to reach the desired temp
- do bed leveling, start printing, etc.

Please move the 
```
M190 S[bed_temperature_initial_layer_single] ;wait for bed temp
```
to where it is actually required.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

StoneLiBambulab, 2023-03-10:
Hi @julie777 ,
The start gcode configuration can be modified by yourself.  you can use asynchronous Gcode command to achieve it.

Thanks!

Stone Li

julie777, 2023-03-28:
I will when I am happy with it. It is taking time to decipher everything that is happening in the shipping startup code because there is no documentation for Bambu Lab gcode.

As for the long wait for the bed to heat,
move the line 

```M190 S[bed_temperature_initial_layer_single] ;wait for bed temp```

from the Heatbed Preheat Section to after 

```;===== prepare print temperature and material end =====```

The section after that is the first time anything is done with the bed so there is no reason to wait for the bed to reach temperature before doing all the nozzle flushing and filament changing. 

With the standard startup code it wastes over 1 minute when the printer is cold just waiting for the bed to heat. With the change the bed reaches temperature at just about the same time as the printer executes the moved instruction to wait for bed temp.