bambulab/BambuStudio #1447: unknown locale causes segfault

cgf1, 2023-03-09:
It looks like, by default, BambuStudio on linux creates a /home/<user>/.config/BambuStudio/BambuStudio.conf file with:
`"language": "en_GB"`

I don't have that language enabled on my system and it caused a segmentation violation before the GUI is displayed:
`Thread 1 "bambustu_main" received signal SIGSEGV, Segmentation fault.
0x00007ffff7eddc3d in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::find(wchar_t, unsigned long) const () from /usr/lib/gcc/x86_64-pc-linux-gnu/12/libstdc++.so.6
(gdb) bt
#0  0x00007ffff7eddc3d in std::__cxx11::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >::find(wchar_t, unsigned long) const () at /usr/lib/gcc/x86_64-pc-linux-gnu/12/libstdc++.so.6
#1  0x0000555556fb475d in wxString::find_first_of(wxUniChar, unsigned long) const (this=0x8, c=..., nStart=0)
    at /usr/src/bambu-studio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/string.h:3233
#2  0x00005555595e889a in wxString::Find(wxUniChar, bool) const (this=0x8, ch=..., bFromEnd=false)
    at /usr/src/bambu-studio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/string.cpp:1652
#3  0x00005555595e79aa in wxString::AfterFirst(wxUniChar) const (this=0x8, ch=...)
    at /usr/src/bambu-studio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/string.cpp:1380
#4  0x0000555556a53c4b in Slic3r::GUI::linux_get_existing_locale_language(wxLanguageInfo const*, wxLanguageInfo const*)
    (language=0x55555ba65bd0, system_language=0x0) at /usr/src/bambu-studio/src/slic3r/GUI/GUI_App.cpp:4395
#5  0x0000555556a5608e in Slic3r::GUI::GUI_App::load_language(wxString, bool) (this=0x55555b8f1d60, language=..., initial=true)
    at /usr/src/bambu-studio/src/slic3r/GUI/GUI_App.cpp:4613
#6  0x0000555556a4190d in Slic3r::GUI::GUI_App::on_init_inner() (this=0x55555b8f1d60)
    at /usr/src/bambu-studio/src/slic3r/GUI/GUI_App.cpp:2145
#7  0x0000555556a3fbeb in Slic3r::GUI::GUI_App::OnInit() (this=0x55555b8f1d60)
    at /usr/src/bambu-studio/src/slic3r/GUI/GUI_App.cpp:2038
#8  0x0000555556a6ed59 in wxAppConsoleBase::CallOnInit() (this=0x55555b8f1d60)
    at /usr/src/bambu-studio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/include/wx/app.h:93
#9  0x00005555595c0e28 in wxEntry(int&, wchar_t**) (argc=@0x55555b6b63b0: 1, argv=0x55555b92d720)
    at /usr/src/bambu-studio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:488
#10 0x00005555595c0f32 in wxEntry(int&, char**) (argc=@0x7fffffff8610: 1, argv=0x7fffffffe078)
    at /usr/src/bambu-studio/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/common/init.cpp:516
#11 0x0000555556a2e79d in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) (params=...)
    at /usr/src/bambu-studio/src/slic3r/GUI/GUI_Init.cpp:64
#12 0x0000555555a689dd in Slic3r::CLI::run(int, char**) (this=0x7fffffffde40, argc=1, argv=0x7fffffffe078)
    at /usr/src/bambu-studio/src/BambuStudio.cpp:459
#13 0x0000555555a803b4 in main(int, char**) (argc=1, argv=0x7fffffffe078) at /usr/src/bambu-studio/src/BambuStudio.cpp:2660`

Hand editing the config file to use "en_US" stops the SEGV.

 - Gentoo Linux
 - Version: 1.50.59 (from git)

StoneLiBambulab, 2023-06-15:
it crash in the CanonicalName.AfterFirst('_') whitch is a function in wxString (wxWidget library).

// get all characters after the first occurrence of ch
// (returns empty string if ch not found)
wxString AfterFirst(wxUniChar ch) const;

I tested the function, including an empty wxString, but they all worked, and I don't  find some problem with the wxString source code either. 
https://github.com/wxWidgets/wxWidgets/blob/d7a98a44ea6c9f814e35db1d1aa869d3b5bf10d1/src/common/string.cpp

I will add some code to avoid empty string which may cause the crash.

Thanks!
Stone Li

WayneManion, 2023-07-13:
This still happens in the 1.7 beta.

jamesawebb, 2024-01-07:
I had the `en_US` locale configured `~/.config/BambuStudio/BambuStudio.conf` but continued to see this error.  In the end I simply generated the `en_GB` locale by uncommenting the appropriate entry in `/etc/locale.gen` the ran `sudo locale-gen`.

xiaguangbo, 2024-05-25:
```
sudo nano /etc/locale.gen
取消 en_GB.UTF-8 的注释，并保存
sudo locale-gen
```

wusar, 2025-03-09:
> ```
> sudo nano /etc/locale.gen
> 取消 en_GB.UTF-8 的注释，并保存
> sudo locale-gen
> ```

Thanks! This works and helps me a lot.