bambulab/BambuStudio #1448: Imported models are not centered

jonfairbanks, 2023-03-09:
**Describe the bug**
Upon importing any model, the object is not centered on the build plate.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio (any version)
2. Click on `Add` in the top menu bar
3. Import your model (stl, 3mf, etc.)
4. The model is not centered. Users must right click the model and press Center

**Expected behavior**
Upon importing onto a fresh build plate, the software should automatically center the object. Users should not be required to manually center objects after import.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version: 1.5.0.60

SaltWei, 2024-01-19:
@Haidiye00 Please follow this issue.
@jonfairbanks Thank you very much for siggestion! And also sorry for seeing this much later

Haidiye00, 2024-01-22:
This issue has been resolved, please wait for version 1.9. thanks.
@SaltWei

SaltWei, 2024-01-22:
@Haidiye00 Please don't close issue before user can get the fixed version!

Haidiye00, 2024-05-06:
The latest version has fixed this issue, please try again. thanks

closed 2024-05-06 by Haidiye00

jonfairbanks, 2024-05-06:
Thanks for the fix!