bambulab/BambuStudio #1450: Add settings to Print Screen

Benderrodrigu3z, 2023-03-09:
I am using a textured plate, well I can hide that section for better view and I could be using a different printer profile on send for this reason these settings NEED to be in the Final Print Popup, else it start and sees that its the wrong plate and you walk back to oh nothing cause plate mismatch

[image]

RyanEwen, 2023-03-10:
I think the reason it's not on the final screen is that changing plates requires reslicing the models and changes the g-code.

Benderrodrigu3z, 2023-03-10:
> I think the reason it's not on the final screen is that changing plates requires reslicing the models and changes the g-code.

Yeah so we NEED to know what we sliced it in, for instance if its on the SD Card and you tell a textured print to print on a high temp because you DON'T REMEMBER and then it times out for wrong plate and you say what the hell plate was it made for?

RyanEwen, 2023-03-10:
Oh I see, you just want to see it on the print screen, not change it on the print screen. Sounds like a good idea to me

YifanWuBambu, 2023-03-13:
Hi @Benderrodrigu3z ,
We need to confirm the requirements with you.
Do you want us to add these 2 changes?
1. add bed type, plate index, printer model(X1C/X1/P1P) on the "Send to printer" dialog 
2. add bed type, plate index, printer model(X1C/X1/P1P) on the printer's screen before printing a model

Benderrodrigu3z, 2023-03-13:
Correct and if its not the default Gcode Start Endcode say what profile the user is calling it for example see pic
[image]