bambulab/BambuStudio #1462: 3D connexion Spacemouse no function in Bambulab studio and Mac

Enderglueck, 2023-03-11:


stefanvaduva, 2023-04-16:
There is a Cura plugin that may be a good starting point for solving this issue: https://github.com/smartavionics/RawMouse I'm also interested to be able to use SpaceMouse in BambuStudio.

pandel, 2023-04-21:
+1 I am so used to my 3D mouse in SuperSlicer, especially for positioning support blockers and such... so please fix it!

jetmansion, 2023-05-17:
Note also that there is a keyboard shortcut for related settings (Command-M), but that keyboard shortcut is the 'minimise' command by default in MacOS. Therefore using that shortcut would just minimise BambuStudio.

leandrolima-nyc, 2023-06-01:
+1 Unfortunately, I was unable to achieve success with BambuStudio. 

However, I can confirm that the desired functionality works flawlessly on OrcaSlicer. Specifically, I utilized the Pro version and successfully incorporated custom commands accordingly, along with increasing the speed of the up, down, right, and left axis movements.

[image]

pandel, 2023-06-10:
Any news on this? I am slicing a dozen parts, which have to be positioned in rather complex ways and normal mouse navigation is a absolutely aweful in this regard! Please, please fix this asap!

jetmansion, 2023-07-13:
I took some time looking in to this and it looks like the problem is caused by the code-signing of the application. The 3Dconnexion drivers is a third party library that needs to be loaded and BambuStudio doesn’t seem to have the correct entitlement set.

The actual code used in OrcaSlicer is identical as that in BambuStudio (at least as far as I can tell through github). But OrcaSlicer is not signed and therefore is not limited by that.

The correct entitlement is actually available in a file in Github (src/platform/osx/entitlements.plist) but maybe this isn't correctly set when code-signing the application?

pandel, 2023-07-14:
@jetmansion YES! You got it! I compiled BambuStudio myself and now my spacemouse is working without issues. So, this is an easy fix. Wait for a new release, compile and be good to go. Many thanks!

jetmansion, 2023-07-21:
@pandel Thanks! Your comment motivated me to do my own build as well. Works great!

Fast-T, 2023-07-25:
1.7 Beta 2 still not working. Please fix this

stefanvaduva, 2023-07-25:
Both PrusaSlicer and OrcaSLicer now have this working. Is there anything else that the community can do to have this working in BambuLab as well? @LiZ-Li-BBL @lanewei120 ?

lanewei120, 2023-07-31:
> @jetmansion YES! You got it! I compiled BambuStudio myself and now my spacemouse is working without issues. So, this is an easy fix. Wait for a new release, compile and be good to go. Many thanks!

we have confirmed if not sign BambuStudio.dmg, it can work(just the same case as built by self)
or if sign the driver with the same codesign, it will also work

but it won't work if using the original driver(unsigned) with the signed BambuStudio

we are still digging into it

lanewei120, 2023-08-05:
already supported in v1.7.2.51, please have a try

closed 2023-08-05 by lanewei120

stefanvaduva, 2023-08-05:
Everything is fine on my end, good job and thank you!

jetmansion, 2023-08-05:
Can confirm that the issue has been resolved. Thanks!

pandel, 2023-08-06:
Works perfectly well :-)! Many thanks!