bambulab/BambuStudio #1463: Octogram spiral pattern always same direction

hutje, 2023-03-11:
Octogram Spiral pattern currently changes direction with the lines that are not part of the complete spiral.
With dual or triple co-extrusion filament you get ugly sections.

It would be nice to have all the lines printed the same direction to get the same pattern on the whole layer.

[image]

hutje, 2023-03-21:
This could be achieved by using the same principles as with the monotonic lines. All lines will be printed in the same direction.
The problem also occurs with concentric and the walls. The walls are printed in the other direction as with concentric pattern.

QingZhangBambu, 2023-04-10:
what gorgeous it is. And how? I'd love to try this, and test your advice.

hutje, 2023-04-10:
> what gorgeous it is. And how? I'd love to try this, and test your advice.

https://github.com/bambulab/BambuStudio/issues/1158#issue-1553898527

With this file, but for the first layer, you select octogram spiral.