bambulab/BambuStudio #1466: Cabin Humidity -far too quick to pop up in Studio

Anthorin, 2023-03-12:
**Describe the bug**
When using Studio in Windows, it only requires the mouse to pass over the Humidity icon for the Cabin Humidity dialog to pop up. It is not very user friendly and it should only pop up either if the icon is clicked or if the mouse hovers over it for say 0.5s. 

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Run Studio and go to Device
2. Pass mouse very quickly over the Humidity icon ( i.e. while moving mouse from say "Calibration" to bottom left of screen. 

**Expected behavior**
Cabin Humidity/Desiccant status help box is only required when icon clicked

**Screenshots**
n/A

**Desktop (please complete the following information):**
 Windows 11

MadMax389, 2023-03-13:
1/2 second delay would be great...

walterwongbbl, 2023-03-15:
we will modify it to click the humidity icon to pop up the humidity description.

walterwongbbl, 2023-08-02:
Hi.
Please use the latest version of Bambus Studio

closed 2023-08-02 by walterwongbbl