bambulab/BambuStudio #1468: Features not present on the Printer are shown in Bambu Studio

suit, 2023-03-12:
**Describe the bug**
When switching to "device" there are features, you see that are not present on the printer (and some of them even marked green as if they were active)

a) Lamp (is shown and always on and green even if you don't have one)
b) AUX Fan (is show, even if you don't have one)
c) Chamber Fan (even if there is one, it is never shown - see also issue #1467)
d) Camera and Settings (even if there is no camera connected and there is a green settings button)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Device"
2. Go to "Status"
3. see for yourself

**Expected behavior**
Only features that are present on the machine should be shown - if this can't be detected automatically, it should be possible to set them manually via some checkboxes to show only stuff that is there

So Lamp, Aux Fan, Chamber Fan and Camera should be greyed out and labeled "not connected"

basically the only feature that does this is the AMS, where it shows "not connected"

[image]

walterwongbbl, 2023-04-06:
can you upload the studio log file

walterwongbbl, 2023-04-06:
If it is P1P, these display states are normal. But some UI needs to be optimized.

Lamp: p1p should not display this UI icon
AMS: We will optimize the UI without AMS
Aux: p1p should not display this UI icon

Thank you for some suggestions. We will continue to optimize.

suit, 2023-04-06:
> If it is P1P, these display states are normal.

"it's not a bug, it's an undocumented feature" :)

what about the Chamber fan? If it is connected, can it be displayed (or at least with a manual option - see also #1467 )

thank you for looking into this and making the interface better

maybe the camera and the light won't be an issue in the near future, since P1P users get the camera and LED for free now anywas

walterwongbbl, 2023-04-06:
Hi.
Is it the cham you installed?

walterwongbbl, 2023-04-06:
This is indeed a issue, and we will open the control of the cham.

suit, 2023-04-06:
yes, the chamber fan is installed

walterwongbbl, 2023-04-07:
Hi.
You can open the control of the cavity fan by modifying the configuration file.
https://github.com/bambulab/BambuStudio/issues/1467
We will carefully consider your other opinions