bambulab/BambuStudio #1469: Speed slidebar constantly displayed

shughey00, 2023-03-12:
**Describe the bug**
once I have activated the speed slidebar by changing the print speed it is constantly displayed in any program as in email..desktop etc. Seems that once the project starts to print if it is displayed it will not or cannot be minimized etc. The only way is to reboot Studio.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. slice project
2. send it to print plate
3. change speeds during or before a print starts
4.stuck visible

**Expected behavior**
Slidebar should never be stuck on visible without anyway to minimize it
[image]
[image]

**Desktop (please complete the following information):**
 windows 64 bit windows 11

walterwongbbl, 2023-03-22:
Hi.
Do you still encounter this issue when using the latest bambu studio version.
Does this issue occur every time?

shughey00, 2023-03-22:
I am not sure about the latest firmware as I run the SoftFever version
which didn’t do it. Could always be a strange issue within windows even
though it’s latest windows 11 update

shughey00, 2023-03-24:
O.K..so it still does this. this is on two different computers running
windows 10 on one and windows 11 on the other. Does it in both Soft Fever
and the latest Bambu sliver version.
It seems to happen if youleave the window open while it finishes a job. It
seems to close o.k. during a job bot oe the job is complete you have to
close out the slicer and restart.
Scott

walterwongbbl, 2023-04-06:
Hi.
We cannot reproduce this problem, but we will try to fixed it.
Improvements will be included in the next release version

shughey00, 2023-04-06:
That’s fine. It is intermittent but has done it on two different devices.
Could be some odd glitch with certain other programs running in background
like say fusion 360 or something like that.

Somerandomgithugguy, 2023-04-17:
I have this problem too and was told it would be fixed in the next release. :)

walterwongbbl, 2023-08-02:
Hi.
Please use the latest version of Bambus Studio

closed 2023-10-13 by walterwongbbl