bambulab/BambuStudio #1470: Cutting an object at a 45 degrees Breaks Dimensions with FIX ANSWER

Benderrodrigu3z, 2023-03-12:
So I raised the green cut out up, rotated it 45 Degrees to plane cut off the corners, rotate it back lower it down and this happens, there is actually a fix, you have to cut it again at a straight line and it resets it back so the 45 degree cut is changing the rotation and scale direction of the object

[image]

Benderrodrigu3z, 2023-03-12:
Cut the Green object at a flat plane height and see what happens to the scale of the object
[AMS Cable Release V2 Cut.3mf.zip](https://github.com/bambulab/BambuStudio/files/10951796/AMS.Cable.Release.V2.Cut.3mf.zip)

zhimin-zeng-bambulab, 2023-10-11:
It has been improved, you can have a try in the latest version(such as V1.7.7).

closed 2023-10-24 by zhimin-zeng-bambulab