bambulab/BambuStudio #1472: Deficiencies of Arrangement Function

KarelBarrel, 2023-03-12:
**Describe the bug**
Instead of arranging the objects in a regular grid, one object is offset or jumps to an unexpected position. Could you please fix the function to behave predictably?

This may seem like a trivial request, but it can often result in confusion, especially when dealing with multiple objects.

**3mf File for This Bug**
There is a picture of what I mean. I am clicking arrange and one object stands out.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Use 3mf 
[Cube_Arrange.zip](https://github.com/bambulab/BambuStudio/files/10952220/Cube_Arrange.zip)

2. Drag the object wherever you want 
3. Press arrange function
4. See: one object stands out

**Expected behavior**
As I have mentioned before - arrange in a regular grid, when it is possible

**Desktop (please complete the following information):**
 - OS: WIN 10
 - Version 1.5.0.60

tangketan, 2023-03-15:
It's because the occlusion region of the cutter and difficult to fix. And it doesn't affect the printing.

tangketan, 2023-04-07:
Our team has managed to solve this problem. Will be in the next release.

tangketan, 2023-08-02:
Solved in latest Studio 1.7.

closed 2023-08-02 by tangketan