bambulab/BambuStudio #1473: [Feature Request] Provide warning and ask for confirmation if object(s) are detected on build plate before starting a new print job

BambuLabDiscord, 2023-03-12:
Before starting a new print job, use the camera and AI to detect objects on build plate. If potential object is detected, give out a warning and only proceed to print if the user either removes the print or confirm no object is present, and presses the Print button. 

This feature will help to avoid starting a new print job when the finished print from the last jog is left on the build plate, either by accident (such as an overnight print that was forgotten to remove the next morning), or in the case of cloud outage when multiple print jobs were sent to the printer by repeated attempts to send over the file.

ThomasFeistel, 2023-03-13:
Hey, I have some wishes and ideas for the slicer. I would like to be able to set up a pause for a layer in the slicer. Also, I would like to have the option to automatically set and regulate the chamber temperature in the slicer.

sprior, 2023-03-15:
For printers which have a door (I think there was some mention that a door sensor exists as well) it would also be useful to pop up a warning if a print is being started when the door has not been opened since the last print finished.  No AI needed.