bambulab/BambuStudio #1475: Device shows "No Printer"

Anthorin, 2023-03-13:
**Describe the bug**
When running Studio, and going directly to "Device", it always shows "No Printer", even though one has been configured.It only shows a printer if you go to "Prepare" first and then back to "Device".

**3mf File for This Bug**
N/A

**To Reproduce**
SStart new instance of Studio.
Go to Device.
This will show "No printer". You can stay here for ever and no printer will appear.
Go to "Prepare" aqnd then back to "Device". Now the printer appears.

**Expected behavior**
Printer should be located in "Device" without having to go to "Prepare" first.

**Screenshots**
n/A

**Desktop (please complete the following information):**
Windows 11

StoneLiBambulab, 2023-03-30:
hi @Anthorin ,
 could help to upload your studio log for me. please refs to wiki 
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks！
Stone Li

SaltWei, 2024-01-19:
@Anthorin Hello, does this issue still exist under your PC environment?