bambulab/BambuStudio #1476: Better Gcode Start Check for QR plate

Benderrodrigu3z, 2023-03-13:
Please update the default gcode to this, it checks for the plate then says warning no plate/QR code, then when you hit continue it resets the Z homing and then Rechecks for the QR code incase the user added it then hit continue, currently if the user does this it SCRATCHES the build plate

;===== machine: X1 =========================
;===== date: 20230130 =====================
;===== reset machine status =================
G91
M17 Z0.4 ; lower the z-motor current
G0 Z12 F300 ; lower the hotbed , to prevent the nozzle is below the hotbed
G0 Z-6;
G90
M17 X1.2 Y1.2 Z0.75 ; reset motor current to default
M960 S5 P1 ; turn on logo lamp
G90
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
M73.2   R1.0 ;Reset left time magnitude
M1002 set_gcode_claim_speed_level : 5
M221 X0 Y0 Z0 ; turn off soft endstop to prevent protential logic problem
G29.1 Z{+0.0} ; clear z-trim value first

;===== heatbed preheat ====================
M1002 gcode_claim_action : 2
M140 S[bed_temperature_initial_layer_single] ;set bed temp
M190 S[bed_temperature_initial_layer_single] ;wait for bed temp

{if scan_first_layer}
;=========register first layer scan=====
M977 S1 P60
{endif}

;=============turn on fans to prevent PLA jamming=================
{if filament_type[initial_tool]=="PLA"}
    {if (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}
    M106 P3 S180
    {elsif (bed_temperature[current_extruder] >50)||(bed_temperature_initial_layer[current_extruder] >50)}
    M106 P3 S255
    {endif};Prevent PLA from jamming
{endif}
M106 P2 S100 ; turn on big fan ,to cool down toolhead

;===== prepare print temperature and material ==========
M104 S[nozzle_temperature_initial_layer] ;set extruder temp
G91
G0 Z10 F1200
G90
G28 X
M975 S1 ; turn on 
G1 X60 F12000
G1 Y245
G1 Y265 F3000
M620 M
M620 S[initial_tool]A   ; switch material if AMS exist
    M109 S[nozzle_temperature_initial_layer]
    G1 X120 F12000

    G1 X20 Y50 F12000
    G1 Y-3
    T[initial_tool]
    G1 X54 F12000
    G1 Y265
    M400
M621 S[initial_tool]A

M412 S1 ; ===turn on filament runout detection===

M109 S250 ;set nozzle to common flush temp
M106 P1 S0
G92 E0
G1 E50 F200
M400
[... 574 lines cut ...]
M109 S{nozzle_temperature_initial_layer[initial_extruder]}
G0 Y0 E1.5 F{outer_wall_volumetric_speed/(0.3*0.6)    * 60}
G0 X60 E1.5
G0 X61
G0 Y25 E1.5
G0 X61
G0 X239 Y1
G0 X179 Y23.5 E4
G0 X239.25 Y1.25
G0 X179.25 Y23.75 E4
G0 X239.5 Y1.5
G0 X179.5 Y24 E4
G0 X179 Y23.5 E2
G0 X239 Y1
;=====END Connect House======

G1 X128.0 Y253.0 Z0.2 F24000.0;Move to start position
G1 E{retraction_length[initial_extruder]} F1800
M109 S{nozzle_temperature_initial_layer[initial_extruder]}

G0 X253 E6.4 F{outer_wall_volumetric_speed/(0.3*0.6)    * 60} 
G0 Y128 E6.4
G0 X252.5
G0 Y252.5 E6.4
G0 X128 E6.4

lanewei120, 2023-03-16:
we have already added similiar logic in previous versions
[image]

Benderrodrigu3z, 2023-03-16:
and yet somehow the nozzle STILL SCRAPPED MY FREAKIN BED, all it needs is a G28 command after you hit continue to not screw up the bed if the user added a plate after seeing the warning, and if they did perhaps check for it and if its the same move on like my code does

Benderrodrigu3z, 2023-03-16:
sorry just upset that I scratched a brand new sticker

lanewei120, 2023-03-17:
do you mean using the newest start-gcode with we released, it still has this issue?

maybe you are using the old 3mf with the start-gcode modified saved in that 3mf?

Benderrodrigu3z, 2023-03-17:
well the current gcode on 1.4.2.13 is what scratched my plate and the fact that there was no safeguard for this from the start makes me less inclined to try the newest gcode when mine fixes the issue

Benderrodrigu3z, 2023-03-18:
> do you mean using the newest start-gcode with we released, it still has this issue?
> 
> maybe you are using the old 3mf with the start-gcode modified saved in that 3mf?

would yall give me a replacement sticker if the new gcode damaged my bed by doing what I said above? your gcode already destroyed one bed

Benderrodrigu3z, 2023-03-18:
also your notes for start gcode clearly says date 2023-01-30, that's over a month and a half ago, there hasn't been an update since then

lanewei120, 2023-03-20:
> > do you mean using the newest start-gcode with we released, it still has this issue?
> > maybe you are using the old 3mf with the start-gcode modified saved in that 3mf?
> 
> would yall give me a replacement sticker if the your new gcode damaged my plate by me doing what I said above? your gcode already destroyed one hight temp plate

please contact our customer support for this issue?
it is better to get reply from them

Benderrodrigu3z, 2023-03-20:
Meaning you you would not and you won’t update the gcode to solve this issue wow okaySincerely,Shane FSent from my iPhone13