bambulab/BambuStudio #1481: Mouse clicks are sometimes ignored on macOS

Hologos, 2023-03-13:
**Describe the bug**
Sometime mouse clicks (both left and right) are ignored. With the right click, the close hint icon jumps (bottom right corner).
It's been happening ever since I got the printer (January 2023).

**3mf File for This Bug**
No related.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio.
2. Right click on objects.
3. See error.
4. Left click on color of object.
5. See error.

**Expected behavior**
Left and right mouse clicks are always registered.

**Screenshots**

https://user-images.githubusercontent.com/1620004/224840186-35f26037-2917-4d3a-a723-6d0050cfa8f2.mov

https://user-images.githubusercontent.com/1620004/224840198-4658793d-47c7-4508-92ee-015664517287.mov

**Desktop (please complete the following information):**
 - OS: macOS 13.2.1
 - Bambu Studio: 1.5.0.60

renssies, 2023-03-16:
Noticing this as well, the buttons next to the plate are often do not register clicks and right clicking a part doesn't reguster the first few times. 

At first I thought it was a delay, but that doesn't seem to be the case. I then thought it was the magic mouse I was using, but it happens on all forms of input, even when using a VPN connection.

**Desktop:**
- OS: macOS 13.2.1
- Bambu Studio: 1.5.0.60

bambu123, 2023-03-17:
Some are really bugs, we'll fix them.
But double-clicking on an item in the Grid View can't let it enter the editing state. This one is not a bug. You should select it first and follow with another click to edit it.

Hologos, 2023-03-17:
> Some are really bugs, we'll fix them.
> But double-clicking on an item in the Grid View can't let it enter the editing state. This one is not a bug. You should select it first and follow with another click to edit it.

In the grid view, I am trying to change assigned filament, not to edit the object's name. I select the object, left click on the filament color and the select box doesn't show up. It shows up on second try. It's the same behavior as other bugs.

bambu123, 2023-04-23:
The problem with right click on plate is fixed.

bambu123, 2023-04-26:
The filament selection problem needs more check. We haven't totally fix it yet.

togeis, 2023-12-09:
Still have the click problem with the current stable version

renssies, 2023-12-09:
I've noticed that the problem is worse when using the Magic Mouse, could it be that the app uses raw data from the mouse and sometimes gets a touch event at the same time and discards the click?

togeis, 2023-12-26:
There are also many mouse click ignore problems in 1.8.2 on MacOS.
- Filament selection 
- align button on the build plate 
- click on objekts to e.g multiply them
- ...

justinwalsh, 2024-01-08:
I am on the latest version and still having click issues when using a Magic Mouse or touchpad. Seems to be worse with the magic mouse. Sometimes, if I click a few times, I can get the plate side buttons to work, and the same goes for right-clicking on models, but it's super frustrating to deal with click not registering.

tablatronix, 2024-06-02:
Same, this is so bizarre

BenoitOClock, 2024-10-22:
same here too

J-charles-C, 2024-11-16:
Same problem with last version.

Thank you

johnbyrneio, 2024-11-23:
I'm having the same problem on Bambu Studio 1.10.1.50 and OrcaSlicer 2.2.0.

I'm running macOS 14.7.1 on a M3 Macbook Air.

Hofmangearsupply, 2024-11-25:
also noticing this, tried OrcaSlicer as well - seemed a bit better compared to Bambu Studio 1.10.1.50 - bought the new m4 Mac mini, is it the Magic Mouse? idk - I'm switching over from windows . Did I make the wrong choice....?

johnbyrneio, 2024-11-25:
> also noticing this, tried OrcaSlicer as well - seemed a bit better compared to Bambu Studio 1.10.1.50 - bought the new m4 Mac mini, is it the Magic Mouse? idk - I'm switching over from windows . Did I make the wrong choice....?

It's not specific to any mouse. I have the same issue with my MX Master 3, Magic Trackpad, and the built-in trackpad on my MacBook Air.

Hofmangearsupply, 2024-11-25:
well that's unfortunate, I have a new track ball mouse coming , hoping that might be the issue.

Hofmangearsupply, 2024-11-27:
I have submitted a support ticket - hopefully this is resolved by the team

Hofmangearsupply, 2025-09-22:
agreed - not fixed as of yet. It's unfortunate that the Mac OS does not seem to be high on the priority list for developers .

robzr, 2025-10-22:
Absolutely still happens to me on various Mac hardware, but only when using Magic Mouse. More details in https://github.com/bambulab/BambuStudio/issues/5403 which may or may not be a duplicate issue of the exact same bug. Really hoping it's fixed someday.