bambulab/BambuStudio #1482: Variable outer perimeter speed causes wall artifacts

Elessar-lad, 2023-03-14:
**Feature Request/Bug**

After slicing an object, the speed preview shows a wide range of variable speeds for the outer walls. The speed changes cause artifacts/defects on the outer wall. 

The 3MF file can be downloaded from [here ](https://drive.google.com/drive/folders/13fh6rYGVR7QaJBQneo5OC8xm1hjidIJY?usp=sharing)

Steps to reproduce the behavior:
1. Import the test piece
2. Choose a filament of choice
3. Layer height is not important
4. Use your default print settings for PLA 
5. Slice the object
6. Switch to the Speed Preview window. See how the closest layers to the solid top infill increase the speed dramatically, even when some of the layers below are not printed as fast. Just happens when close to the top solid infill. See how the outer wall speed constantly varies throughout the height of the piece. 
7. Print the piece. 
8. Inspect the outer wall quality.
9. See how the external defects match the layer where the speeds increased as mentioned in Step 6. 

**Expected behaviour**
The outer wall speed should be constant thought the print (except when there are small features or the layer time is below the set value) regardless if the printed layer is the top solid infill or close to it. If you use these settings, 100mm/s outer wall, 8 sec layer time, 2 walls, 30%infill, honeycomb pattern, and using Bambu PLA matte see how the object is printed with constant outer wall speed. Changing any of the suggested values (number of wall, infill percentage, etc) causes the part to print with variable outer wall speed, which in turn causes outer wall defects, like bulging or uneven extrusion (z banding).

**Screenshots**
Find them [here](https://imgur.com/a/n64yy3s)

**Desktop (please complete the following information):**
 - Windows 11
 - Bambu Studio 1.5.0.60

LordBoos, 2023-03-14:
I've also noticed this, some prints combined with some filament brands have very bad artifacts because of this. It's also very prominent when using adaptive layer height, which also creates this outer wall speed differences.

QingZhangBambu, 2023-03-20:
you can close the cooling option to get constant  out wall speed
![Uploading image.png…]

QingZhangBambu, 2023-03-20:
[image]

closed 2023-04-06 by QingZhangBambu