bambulab/BambuStudio #1483: ABL data

ab3dgeek, 2023-03-14:
Would it be possible to have access to the bed-leveling data?
(either like on klipper as a height-map image or just the raw height-data in a spreadsheet)
I would really like to see if/when/how much the already bowed (...) print-bed deforms and when it is time to change it.

Not one is flat!
I'm not expecting 'perfect' flatness, but I've seen others talk about anything like 0.2-1mm from the middle to the corners which is just bad!

Auto bed-leveling works and most first layers are just fine but that doesn't make the printed objects bottom flat.
Every bigger print I did just noticeably rocks when put on a flat surface.
They are bowed from new out of the box. All of mine are dipped in the middle and goes higher on towards each corner.

My x1c's are not as bad as the p1p's but all are pretty bad compared to my Vorons (cheap kit! not LDO).
Surprisingly even any of my 10x Ender 3s (some of which is like 6-7 years old by now!) produce flatter bottoms.
I'm going to measure the actual differences on my beds with a digital indicator as soon as I have the time, but it would be way quicker to see the data from ABL at first.
(before someone says/asks: no this is not a warping issue.)

suit, 2023-03-14:
that would be really helpful - if there are noteable lowspots on the heated bed, they can be shimmed with adhesive, temperature resistant aluminium tape

but for this you need to make multiple prints (full sized over the entire bed) with different build plates to find those low spots, since the plates themselves have slight variations in flatness

this is extremely wasteful - if we can just extract a heightmap (greyscale image) for the calibration data, this would be way easier

adamfilip, 2023-03-14:
I agree, please output a CSV file + a Image Heat Map to the Media Folder in Bambu Slicer for each job

noswall, 2023-03-18:
This would be very helpful. A heatmap and raw data would be great.

NairoDorian, 2023-03-27:
YES WE NEED THIS SINCE DAY 1