bambulab/BambuStudio #1485: [Feature request] Resume partial printing

mainmind83, 2023-03-14:
When you print several objects on the same plate and the printing pauses when detecting spaghetti, for example, possibility to skip one of the objects and continue with the rest that has no problems
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
surely it is difficult to implement a solution from the generated code, but is there any possibility? It would help not to waste filament and resume partial prints

ev1ls33d, 2023-03-16:
Good that I searched before posting.
I wanted to propose exactly the same. A function in BambuStudio to enable skipping of failed objects.

Maybe, when spaghetti is detected, the printer uses the lidar and checks the current layer. If one object is not at the expected height, it will wipe its nozzle and skip it after resuming (and the user has cleaned the spaghetti).

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei

mainmind83, 2023-08-02:
Great @SaltWei ;D

[image]