bambulab/BambuStudio #1486: Auto Delete old Timelapse on SD

adamfilip, 2023-03-14:
When SD card runs out of room and you still want to capture a timelapse.
enable auto delete of oldest timelapses so it can record a new one.

Also enable a Heart Icon to save certain ones, protection so those specific ones dont get deleted automatically.

sodachen22, 2023-03-16:
In the original design, the timelapse video is used to be downloaded and shared with friends. So:
1. It is supproted to download and deleted timelapse from Studio and Handy.
2. The timelapse enabling option is available in every printing job, the user can decide to enable timelapse or not.
3. It is not the same as video recording file which is deleted automatically when there is no enough free room

wsegatto, 2025-10-07:
> In the original design, the timelapse video is used to be downloaded and shared with friends. So:
>
> 1. It is supproted to download and deleted timelapse from Studio and Handy.
> 2. The timelapse [...]

Is there any documentation confirming that older videos get automatically deleted (cycled through) when the SD card is full?