bambulab/BambuStudio #1487: Bambu Studio 1.5 does not load on Windows 11

frasier2k, 2023-03-14:
Bambu Studio 1.5 does not load on Windows 11. When starting the app nothing happens. No splashscreen. 
Previous Version 1.4.2 is running w/o issues.

Tried to run as Administrator, but no success. I also tried another computer with Win11 and it was working.

lanewei120, 2023-03-16:
thanks for reporting this issue
1. could you help to upload the log in C:\Users\your_user_name\AppData\Roaming\BambuStudio\log?
2. also, you can try to rename C:\Users\your_user_name\AppData\Roaming\BambuStudio to C:\Users\your_user_name\AppData\Roaming\BambuStudio.bak, and then have a try to check whether it is related with previous data
if yes,  could you also help to zip this folder to us for a check?

frasier2k, 2023-03-18:
After a clean installation and trying to start the app the log folder was still empty.

I checked the installation of "vcredist2019_x64.exe" C++ Runtime and repaired it. Now it is working.

closed 2023-03-18 by frasier2k