bambulab/BambuStudio #1488: Ää Üü Öö ß -  German Umlaut's not usable (3D Text Tool)

Popoboxxo, 2023-03-14:
**Describe the bug**
It is not currently possible to use "umlauts" - Ää Üü Öö ..." -in the 3d text tool..
[https://en.wikipedia.org/wiki/Umlaut_(diacritic)

And the German spare is unfortunately full of them.

**3mf File for This Bug**

All 3mf Files

**To Reproduce**
Steps to reproduce the behavior:

1: Start Bambu-Studio
2: Add an random 3D-Object
3: Use the 3D text tool
4: Type some "normal" letters
5: Now try to add german "umlauts"
6: ( if umlauts are used, the text can also no longer be moved )

**Expected behavior**
Text's like "test ä # text" can be generated
Text's like "test Ä # text" can be generated
Text's like "test ö # text" can be generated
Text's like "test Ö # text" can be generated
Text's like "test ü # text" can be generated
Text's like "test Ü # text" can be generated
Text's like "test ß # text" can be generated (also an special "character)

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 22H2 Build 22621.1265
 - Version: 1.5.0.61

ChrMaass, 2023-03-14:
That's exactly what I was about to report. Annoying, because I also have an umlaut in the name and just wanted to print a few sachen 😅

Popoboxxo, 2023-03-14:
Worst Case :D

zhimin-zeng-bambulab, 2023-04-18:
It has been fixed in v1.6 released on github.

closed 2023-04-18 by zhimin-zeng-bambulab