bambulab/BambuStudio #1489: Ability to stop individual objects during a print, especially in case of failure

39ckjdu45, 2023-03-14:
Problem: Long prints with multiple parts can have a part fail, and the whole print usually needs to be canceled or there is a risk that other parts will fail due to the spaghetti stuck to the nozzle as well as being dragged around the plate. 

Solution: Allow ability *on* the printer to cancel individual objects both while the print is going and while the print is paused. 

Alternatives: The alternative is the let the printer spaghetti plastic about the plate and hope it doesn't ruin another object or to cancel the whole print.

mainmind83, 2023-03-14:
Same feature here: https://github.com/bambulab/BambuStudio/issues/1485

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei