bambulab/BambuStudio #1490: Slicing cuts off part of object

julie777, 2023-03-14:
# Describe the bug
When I slice an object it gets cut off and only the bottom is shown.

[image]

[image]

And the layer slider only shows 11 layers.

Choosing a different print profile, such as .16 optimal @BBL P1P changes the number of layers in the slice, but still cuts it off at 2.12mm

I have restarted the computer, bambu slicer, sent the model directly from Fusion 360, loaded it from a saved 3mf file. It always fails slicing.

# 3mf File for This Bug

[testbin.zip](https://github.com/bambulab/BambuStudio/files/10973271/testbin.zip)

# To Reproduce
Steps to reproduce the behavior:
1. Load 3mf file
2. slice

# Expected behavior

It should slice correctly like it does in Prusa Slicer.

[image]

# Desktop (please complete the following information):
 - OS: Windows 10
 - Version: Internal Version 1.5.0.59

SaltWei, 2023-03-16:
[image]
[image]

Not bug. But just different settings with prusa.
Bambu's default wall generator is classic not arachne.
The default generator is not arachne, becuuse it's has a lot of bug at the same time.
For example, arachne can't slice the model rightly.
[ccc.zip](https://github.com/bambulab/BambuStudio/files/10989552/ccc.zip)

julie777, 2023-03-19:
So when I see this behavior I need to change the perimeter generation to classic?

I do think it is a bug if the slicer can't slice the model correctly and provides no information that there was a problem.

SaltWei, 2023-03-20:
> So when I see this behavior I need to change the perimeter generation to classic?

The default perimeter generation of BS is classic, and switching to arachne can heal this model.

SaltWei, 2023-03-20:
But for some other model, arachne can't heal rightly, for example [ccc.zip](https://github.com/bambulab/BambuStudio/files/10989552/ccc.zip).
Classic and arachne are suitable for different scenarios sometimes, so the perimeter generater is a switch.

SaltWei, 2023-03-20:
> It should slice correctly like it does in Prusa Slicer.

No, prusa also has this problem.