bambulab/BambuStudio #1491: [Feature Request] Allow renaming printer in LAN Only mode

BambuLabDiscord, 2023-03-14:
Allow users to rename their printers in LAN Only mode. Currently one can rename their printer(s) in the cloud, but understandably the names will not be carried over to LAN, and there isn't a way to rename the printers in LAN. Being able to have shorter printer names will make it easier to identify printers in a multiple printer setup in LAN mode.

jbeardon, 2023-03-16:
+1000 !!!
 If you have more than one printer they are a nightmare to manage in Lan only mode (which, incidentally I have been FORCED to enable because MQTT has been killed on the P1P unless you are in Lan only).

StoneLiBambulab, 2023-04-03:
Hi @sodachen22 ,
    it requires printer firmware support.

Thanks!
Stone Li

DHernan, 2023-08-18:
Adding my vote for this after the cloud fiasco

Darthmonkey, 2024-01-02:
As more people get more Bambu printers, this is becoming more of a hot topic, in LAN mode my P1S name has 'P1P' in it, and my P1P has 'P1S' in the name, not confusing at all.

SoCuul, 2024-01-22:
It would be nice if we could do it just locally, with the name being stored in the bambu studio settings.

rspeed, 2024-02-19:
We have a P1P and P1S in the lab I manage. For whatever reason, their serial numbers (and therefore their names in LAN-only mode) start with "01S" and "01P", respectively. Note how they're the opposite of what you would expect. If I didn't know better, I would think Bambu Lab did this intentionally.

amdavidson, 2024-03-16:
Even if Bambu won't support this in firmware, being able to alias locally would still be a big improvement.

SoCuul, 2024-05-08:
@StoneLiBambulab would it be possible to locally alias printers?

vordenken, 2024-06-05:
Any updates on this? Would be a really nice feature - even more interesting now that BambuStudio can control multiple printers at once.

rspeed, 2024-07-29:
Naming LAN-only printers is now natively supported in the latest P1 firmware. Can anyone confirm if that's also the case for X1 and A1 printers?

metawops, 2025-02-24:
Can't confirm for X1C. Upgraded to Firmware 01.08.02.00 and the printer name in Bambu Studio returns to the default 3DP-xxx-xxx name when switching to LAN only mode. 😞

Hammerfest, 2025-02-26:
Here we are, just a short time hop from 2 years since this thread's creation, and still cannot rename printers in LAN only mode.

hayden-servaenergy, 2025-05-06:
Renaming X1C in LAN mode is still not possible with 1.09.00.01