bambulab/BambuStudio #1493: Z layers cursor detach

Sergiuli, 2023-03-15:
**Is your feature request related to a problem? Please describe.**

When you use the Z axis cursor with the mouse you are looking at the object and often the mouse trajectory does not accurately follow the verticality of the cursor. This determines that during the observation you find yourself with the plane rotating because you have lost contact with the cursor and this is very annoying as well as a big waste of time.

**Describe the solution you'd like**

when scrolling layers by moving the cursor up and down with the mouse it should not detach even if the mouse does not follow the vertical trajectory

Thank for considering this feature proposal

lanewei120, 2023-03-16:
hello, thanks for reporting this,
could you give a short video on the details?

Sergiuli, 2023-03-16:
Sorry but i don't know how to make screen video but i will try to explain myself better. 
When you scroll through the various layers of the object to be printed with the Z axis cursor on the right of the screen, you must be careful not to move the mouse pointer from the vertical axis of the cursor otherwise you will lose control of it. 
This happens because when you scroll your gaze is focused on the object and the layer and not on the cursor. 
Once the cursor is hooked, it should be kept controlled by the mouse until the button of the same is released.
Thank you very much for taking my suggestion into consideration and if I haven't been clear enough, please let me know and I will try to make the video you asked for in some way.

Sergiuli, 2023-03-23:
Ok, thanks

closed 2023-03-23 by Sergiuli