bambulab/BambuStudio #1496: Network Timeout is too short

julie777, 2023-03-15:
**Describe the bug**

When I select print, the print dialog opens and it starts refreshing printer status. 
It almost always times out.
- sometimes it updates the information immediately after printing the timeout message
- other times I have to select refresh. When I do refresh returns immediately and the status is updated.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
I expect the print dialog to be shown and the printer status to be updated immediately without any delay for refresh.

The network logic is not implemented very well. 
- when I select print it knows which printer because I only have one
- when I am not printing and go to devices it doesn't select my one printer, instead I have to select it
- when I select my printer on the device page it takes a long time to connect
- if I am on the device page and connect to my printer and then immediately go to preview and select print it opens the print dialog and then starts waiting to refresh, even though it is already connected. And then it will timeout trying to refresh.
- opening a second instance of Studio will always crash when any network activity happens

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

julie777, 2023-03-31:
@StoneLiBambulab 
Just to clarify,
- That might be true when connecting to the cloud but connecting via LAN mode doesn't incur a cost.
- If I am on the device page and it is updating status, which I can see happening then it is currently connected to the printer. Then I select print and it opens the print dialog and times out.

I have submitted a Bambu Lab support request on this item, so if you are a Bambu Lab employee I have provided much more information through that channel.

julie777, 2023-03-31:
I just realized that our most recent exchange has nothing to do with the topic of this issue.
When a print starts, it opens the print dialog that says it is synchronizing printer status.

This always times out. 

I can then refresh and it will update and allow me to send the print.

It is the fact that it always times out that I was reporting.

Also, I have written a small program that uses MQTT to 
- connect to the printer
- subscribe to messages
- ask the printer to report status
- show the status when received.

From the time my app requests a connection to the time it prints full printer status is always .1 seconds. This tells me that something in the bambu studio communication code is not working optimally.

I have run the Bambu Studio test both with the internet connection to allow it to succeed in all the unnecessary cloud connections and with no internet. It always times out in the print dialog.

StoneLiBambulab, 2023-04-03:
Hi @julie777 ,
   Studio needs to synchronize the status from the printer when sending a print job, including information such as whether it is printing and whether it needs to be upgraded.  If the printer does not send the status message, studio reports a timeout. This often happens when the printer network is not working well. Check the printer's network, such as signal strength maybe helpful, or put your printer closer to your router.

Thanks!
Stone Li

julie777, 2023-04-04:
I have checked my network and it is fine. I also wrote an app that connects to the printer with MQTT and subscribes to topics. It then requests the printer to post all status.
The app prints every message it receives from the printer.

I just did a test for you. I ran my app and the time stamps will show you how long it took to connect and how long for the info request to return. It took a total of 4.3 seconds from the request to connect to the time it received the info back.

[bambu.log](https://github.com/bambulab/BambuStudio/files/11143595/bambu.log)

Then I started bambu studio, loaded a model and clicked on start print. I used a stop watch and it took
16.51 seconds before the sync timed out message appeared. Then I just kept watching and eventually it updated the status and was ready to send. This took 
17.95 seconds more after the sync error occurred.
So in all it too Bambu Studio 34.47 seconds to connect and get status from the printer.

Here is the network log from that test.
[debug_network_Mon_Apr_03_18_34_02.log](https://github.com/bambulab/BambuStudio/files/11143627/debug_network_Mon_Apr_03_18_34_02.log)

I am not looking at your source code, but one thing that stands out to me is the request appears to be sent before subscribing which could cause the info response to be missed.
at line 38
```
[2023-04-03 18:34:42.710654 -06:00] [I] [T 29976]: local send message topic=device/01S00A292501218/request, payload={"info":{"command":"get_version","sequence_id":"20002"}} 
[2023-04-03 18:34:42.722825 -06:00] [I] [T 5420]: subscribe topic:device/01S00A292501218/report success
```
I cannot be sure because the responses are not logged. It does send the request again later.

JericoPablo, 2025-01-19:
o still have the same issue. @StoneLiBambulab  why dont u answer to this anymore? Why isnt this being fixed?

sdthamburg, 2025-06-29:
X1C is dropping connection during print. Switching from showing video & current printer status to showing "no printer".