bambulab/BambuStudio #1497: Bambu Studio crashes when using LAN only mode and cloud is not accessible.

julie777, 2023-03-15:
**Describe the bug**
I frequently have Studio hang. Much of my troubleshooting is in #1361 
In my log there are many attempts to contact the Bambu cloud, which is not accessible on my network.

[debug_network_Wed_Mar_15_11_10_06.zip](https://github.com/bambulab/BambuStudio/files/10982974/debug_network_Wed_Mar_15_11_10_06.zip)

BTW, when it hangs there is never a debug log.

**Expected behavior**
Bambu Studio should have a LAN only mode so that it can work well with printers in a LAN only setup and not try to connect to outside servers.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version Internal Version 1.5.0.59

walterwongbbl, 2023-03-16:
Hi.
Please close the firewall and try again.

julie777, 2023-03-17:
The crash appears to have been a different issue. I can print fine on a LAN without internet access. There are errors in the logs about trying to access bambulab.com but it works fine. 

There are many issues with communication to/from the printer and I am working on them in #1361.

closed 2023-03-17 by julie777