bambulab/BambuStudio #1498: Lücke zwischen Brim

Ic3breaker, 2023-03-15:
Wenn man die X-Y Konturkompensation auf einen Wert von -0,2 und / oder die Elefantenfußkompensation >0,1, dann ist ein Spalt beim Brim. Der Brim verliert somit seine Funktion, weil er nicht mehr mit dem Bauteil verbunden ist...

suit, 2023-03-16:
> If you set the X-Y contour compensation to a value of -0.2 and / or the elephant foot compensation >0.1, then there is a gap at the Brim. The brim thus loses its function because it is no longer connected to the component....
Gap between edge and object is at 0

the gap between the brim and the object is intentional, it should not directly be merged with the object - if you don't want this, just remove the brim-object gap

https://wiki.bambulab.com/en/software/bambu-studio/auto-brim#brim-object-gap

QingZhangBambu, 2023-03-20:
as @suit said, you can remove the brim-object gap to solve the issue.

closed 2023-03-20 by QingZhangBambu