bambulab/BambuStudio #1499: Multiple Text Fails on the Newest Update 1.5.0.61

Benderrodrigu3z, 2023-03-15:
Nan? how is an object I just created at Nan when I clicked on the surface?

[image]

[Screen Recording 2023-03-15 at 2.04.41 PM.mov.zip](https://github.com/bambulab/BambuStudio/files/10983800/Screen.Recording.2023-03-15.at.2.04.41.PM.mov.zip)

https://user-images.githubusercontent.com/48887257/225417391-3bc8edb9-a92e-47b6-b4a3-d69382466c2e.mov

Benderrodrigu3z, 2023-03-15:
another issue in the Screen Recording ZIP, its a Zip cause the video is 28mb

Benderrodrigu3z, 2023-03-15:
Object Gets NAN how here:

Uploading Screen Recording 2023-03-15 at 2.21.15 PM.mov…

Benderrodrigu3z, 2023-03-15:
How to Nan an Object:

https://user-images.githubusercontent.com/48887257/225420213-88c2cf1a-c1d7-4130-988f-cfadcb30f041.mov

Benderrodrigu3z, 2023-03-15:
right when you click bam NAN, can you at least make it use a different filament color so we can see that its laying the object?

https://user-images.githubusercontent.com/48887257/225421047-0c9028c5-474e-4ff0-890c-c035d9c71e56.mov

Benderrodrigu3z, 2023-03-15:
https://user-images.githubusercontent.com/48887257/225423299-56d549eb-8062-4fe7-852d-0187437ebddb.mov

zhimin-zeng-bambulab, 2023-03-16:
@Benderrodrigu3z  Could you share your 3mf file, it will help us to solve this problem, thanks.

Benderrodrigu3z, 2023-03-16:
[Slot_BLANK.3mf.zip](https://github.com/bambulab/BambuStudio/files/10991841/Slot_BLANK.3mf.zip)

zhimin-zeng-bambulab, 2023-03-17:
Thanks, I can reproduce this problem, we will fix it in the next version.

Benderrodrigu3z, 2023-03-17:
Sounds great thanksSincerely,Shane FSent from my iPhone13

zhimin-zeng-bambulab, 2023-04-18:
It has been fixed in v1.6 released on github.

closed 2023-04-18 by zhimin-zeng-bambulab