bambulab/BambuStudio #1501: Filament profiles deleted after update :(((

MartinM8266, 2023-03-15:
The update of the last bugfix deletes the custom-filament profiles!

is this really neccessary???? EXTREMLY annoying!

MartinM8266, 2023-03-15:
Filament profiles are stored in C:\Users\XXXX\AppData\Roaming\BambuStudio\user\XXXX.
These are inherited profiles from the standard profiles. Only changes in comparison to the standard profiles are beeing stoered in the json-files.
Managed to recover the filament profiles, but there should be any kind of warning or automatic reimport at least.

Benderrodrigu3z, 2023-03-16:
More like try to delete them and they keep redownloading from the backup, they won't deleted

cmguo, 2023-03-17:
Please turn on 'Sync user presets'.

Benderrodrigu3z, 2023-03-17:
Where is this setting on Mac?Sincerely,Shane FSent from my iPhone13

bambu123, 2023-03-17:
[image]

Benderrodrigu3z, 2023-03-17:
So mine is missing the Associate file types on Mac, also mine was set to 30sec which was causing my issue cause I was closing without waiting, you can close this cause once I waited longer it workedSincerely,Shane FSent from my iPhone13

bambu123, 2023-03-21:
My picture is from the Windows system. I only mean the ‘Auto sync user presets'.
@MartinM8266 Had you turned on the switch?

MartinM8266, 2023-03-22:
yes.sure.

______
Martin Mostböck / +43 6991 3415453 / ***@***.***
von Android gesendet.

bambu123, 2023-03-28:
> Filament profiles are stored in C:\Users\XXXX\AppData\Roaming\BambuStudio\user\XXXX. These are inherited profiles from the standard profiles. Only changes in comparison to the standard profiles are beeing stored in the json-files. Managed to recover the filament profiles, but there should be any kind of warning or automatic reimport at least.

So what did you do to recover the filament profiles? Were the files under the 'user' directory lost?

dkern, 2023-04-04:
I had the same issue after update, all my custom profiles are missing now. I have found the files in my profile folder, but I have no idea how to restore them.

bambu123, 2023-04-04:
@dkern Turn off and then on 'Sync user presets'.

bambu123, 2023-04-06:
It occurs after updating the network plugin after updating BambuStudio. Restarting BambuStudio may help.

bambu123, 2023-04-23:
Fixed in version 1.6, please have a try.

closed 2023-04-23 by bambu123