bambulab/BambuStudio #1502: Change printorder per plate

ChrMaass, 2023-03-16:
I try to optimize my prints and print as many parts at once. Sometimes it makes sense to print multicolored parts in order by object, but for other parts of the same color it makes more sense to print by layer, because then you can make the plate fuller.

It would be great if you could select the print order depending on the print plate and not only globally for the whole project. Or at least override it for individual plates.

lanewei120, 2023-03-17:
we have already provided this option for each plate

could you try it
[image]

ChrMaass, 2023-03-23:
Thanks a lot, I have overseen it.

closed 2023-03-23 by ChrMaass