bambulab/BambuStudio #1504: Provide automation friendly MQTT topics

sprior, 2023-03-16:
Currently to use the MQTT status topics/payloads for home automation integration it is required to subscribe to and parse messages that happen every few seconds just to decide which messages represent a major status change (power on, print start, print end).  This is because the same topic is used for all and the nature of the event is encoded in the message body.

**Describe the solution you'd like**
A far more automation friendly MQTT topic structure would allow the type of event to be specified in the topic path.  For someone trying to catch print start and print end messages this would reduce the subscription parsing workload from every few seconds to just two.

**Additional context**
Moonraker just made exactly the kind of commit I am requesting: https://github.com/Arksine/moonraker/commit/e3bcfb5ea0a3256bdbe19855d20c4431b57cd398

user897943, 2023-03-16:
+1 absolutely! Separation into single topics would be way more efficient. For instance, power status would be appearing on every update during a print causing overheads but that's unnecessary, if a print is ongoing it's safe to assume the printer is on. Yet, by being in a single message it's included, that's unnecessary bytes being transmitted on the network. Separation into single topics would mean updating only what changes and when it changes. Good suggestion.

LayerCakeMakes, 2023-03-17:
+1 this just makes sense to me

FlyingHippo, 2023-03-27:
+1
But https://github.com/bambulab/BambuStudio/issues/1536 has higher priority for me

kylek29, 2023-03-30:
+1 Sensible request.

Nicoautoxp, 2023-04-05:
+1 for me too

NPetersenDK, 2023-04-30:
+1 - love more MQTT Integration <3

hubertron, 2023-12-17:
Would love to see this as well

cbenhagen, 2024-01-06:
Please upvote the issue by giving it a thumbs up instead of leaving +1 comments. Issues can be sorted by reactions. And even more importantly you don't send a notification to everyone subscribed to this issue.