bambulab/BambuStudio #1506: Switching Window Cam-Stream in Bambu Studio 1.5.0.61 between 2 Monitors

FOR95, 2023-03-17:
Hello!

Since i used Bambu Studio with my new X1C, i always get this anonying bug
that the Bambu window which shows actually the Cam-Stream stops while 
i move it between the monitores. I always need to click again on the play 
button to start the stream.

I tried it with Live and 720/1080p, with and without SD-Card.

Windows 11 - 22H2 - 22621.1413

Greetings,

Fabian

bambu123, 2023-03-23:
@FOR95 Is that so?

FOR95, 2023-03-23:
@bambu123 Hey ! Sorry for my late reply.

Yes you are right, after around 5 - 10 Seconds the Cam show up again. But maybe we could do it with less
delay ? For normal Users it will be not "normal"

Thanks and Greetings,

bambu123, 2023-04-23:
Fixed in version 1.6, please have a try.

closed 2023-04-23 by bambu123

FOR95, 2023-04-23:
> Fixed in version 1.6, please have a try.

Hey Bambu123!

I try it, much better but its still laggy and ever 5 - 8 change from one to the other desktop it stops.

Greetings,