bambulab/BambuStudio #1507: Start printing faster - performance improvement idea

julie777, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
It is just a response to the general complaint that it takes the Bambu Lab printers too long to get ready and start printing.

**Describe the solution you'd like**

This is based on the current communication times between Bambu studio and the printer. I don't know whether it would make much difference if the model transfer to the printer actually happened at >500Mbits/sec

Given that Studio already has the ability to talk to the printer and perform actions directly, when I select print Studio could
set the bed temp based on the model
set the print head based on the model
then send the model to print.

The slowest part of the required actions that must be performed before printing is the heating time for the hot end and the bed - especially the bed. If I am transferring a 5Mb model and it is slow, which it usually is, then the printer might have a minute or more to start heating before it could start processing the gcode.

**Describe alternatives you've considered**
I have already modified my start gcode to add various optimizations such as not waiting for the bed to be at temp before starting to heat the hot end.

**Additional context**
Add any other context or screenshots about the feature request here.

NairoDorian, 2023-03-27:
You can do it yourself by changing the start gcode, I did it myself and posted everything on Printables

julie777, 2023-03-31:
I have done it myself. I just posted it here to help new Bambu Lab printer users.

Stefan-Hanke, 2023-04-01:
Interesting - can somebody provide a link? I fail to conjure up the right search terms...

julie777, 2023-04-01:
@Stefan-Hanke 
https://www.reddit.com/r/BambuLab/comments/125700m/would_you_like_to_safely_shorten_printer_startup/