bambulab/BambuStudio #1509: Why can't Studio jump to the device page immediately after print is sent?

julie777, 2023-03-17:
**Is your feature request related to a problem? Please describe.**
I hate waiting. Also, this delays the turn on of the camera.

**Describe the solution you'd like**
Jump to the device page as soon as the send is completed successfully.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Merlwynd, 2023-03-17:
Mine does....After I send the file to print, Bambu Studio automatically goes to the device page and then I start the camera (doesn't start automatically unless I have already sent a print earlier in that instance of Bamby Studio).....Maybe there is a setting involved?...although I don't remember one...

julie777, 2023-03-19:
Mine does a 3 second count down before switching.

StoneLiBambulab, 2023-03-20:
Hi @julie777 , 
just give the users a moment to confirm that the task has been sent.

Thanks!

Stone Li

closed 2024-01-29 by StoneLiBambu

bolsoncerrado, 2024-01-29:
@StoneLiBambu  did you close this because "Print file" already does this or you added the behaviour to "SEND TO PRINTER" too?

I ask to avoid opening a new FEATURE REQUEST duplicate.

StoneLiBambu, 2024-01-31:
"Print plate" will send the current plate printing job to printer and start to print and "Send" will only send the xxx.gcode.3mf to printer, you can start the print job on printer screen or MicroSD card on device Page.
[image]

[image]

Thanks!
Stone Li（旺仔）