bambulab/BambuStudio #1510: [Feature Request] Aux Fan Speed based on Bridge Length and/or Overhang Angle

Photogad, 2023-03-18:
**Is your feature request related to a problem? Please describe.**
I don't know if it's possible with Bambu Firmware but it irks me that there's no way I can find to adjust Aux Fan cooling speed based on bridge length or overhang angles. Being able to adjust this to 100% on the fly would allow for better bridging and overhangs. As it is now, it seems it can only be set to 1 constant speed (default of 70% for PLA)

**Describe the solution you'd like**
Settings to adjust Aux Fan Speed based on Bridge Length and/or Overhang Angle

**Describe alternatives you've considered**
none

**Additional context**
none

Manfredo1952, 2023-03-23:
This request coincides with my problem to find suitable settings for bridging.  I am trying to work on a test file like this (https://www.printables.com/model/109216-bridge-test-fast-low-cost-ender-3) buit keeping getting bad resulst when viewing the printed object from below. Have tried different speeds etc, to no avail. Would appreciate feedback   
[image]

Photogad, 2023-05-14:
bump
@StoneLiBambulab this would make printing on this printer a LOT better

nirurin, 2023-07-28:
This would be great... Why doesn't the aux fan already have this control? I assumed it would be controllable per layer/feature same as the part cooling fan. Otherwise what's the point.

eckozen84, 2023-08-09:
I agree, it just adds so much more tuning flexibility to cooling the model for overhangs and bridges (much like the part cooling fan). I don't understand why this feature wasn't included from the start.

thepowersource, 2023-09-09:
All fans should really be incorporated into the slicing options. By layer, overhang angle, cabin temp etc etc.

christabone, 2024-12-05:
If anyone is still in search of a solution for this, I've written a Home Assistant automation that lets you control the AUX fan speed by layer:

https://github.com/christabone/aux_fan_by_layer

It's not as fancy as detecting bridges or overhangs, but if you take note of the layers in the slicer, you can input them into this automation for better cooling control.