bambulab/BambuStudio #1511: BambuStudio 01.05.00.61 cannot connect to the printer

0rsa, 2023-03-18:
**Describe the bug**
BambuStudio cannot connect to the printer

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade from Bambu_Studio_win-v01.05.00.60.exe to Bambu_Studio_win-v01.05.00.61.exe
2. Open BambuStudio
3. Click on device tab
4. You cannot get information from the printer
5. If you try to launch a print, the filaments are never resolved in the selection box

**Expected behavior**
BambuStudio is sync with the printer

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11 Professionnel 
 - Version 21H2 22000.1574

**Workaround**
- Re-install Bambu_Studio_win-v01.05.00.60.exe, it works again.

0rsa, 2023-03-18:
Some logs

01.05.00.60:
`[2023-03-18 05:52:53.814731 +01:00] [I] [T 58176]: init log
[2023-03-18 05:52:53.815070 +01:00] [I] [T 58176]: init_log
[2023-03-18 05:52:53.815090 +01:00] [I] [T 58176]: Agent: set_cert_dir_name, dir=C:\Program Files\Bambu Studio\resources/cert, name=slicer_base64.cer
[2023-03-18 05:52:53.815144 +01:00] [I] [T 58176]: set_on_server_connected_fn
[2023-03-18 05:52:53.815150 +01:00] [I] [T 58176]: set_on_printer_connected_fn
[2023-03-18 05:52:53.815151 +01:00] [I] [T 58176]: set_get_country_code_fn
[2023-03-18 05:52:53.815152 +01:00] [I] [T 58176]: set_on_local_connect_fn
[2023-03-18 05:52:53.815152 +01:00] [I] [T 58176]: set_on_message_fn
[2023-03-18 05:52:53.815153 +01:00] [I] [T 58176]: set_on_local_message_fn
[2023-03-18 05:52:53.815186 +01:00] [I] [T 58176]: set host to https://api.bambulab.com/v1
[2023-03-18 05:52:53.815188 +01:00] [I] [T 58176]: region update iot = https://api.bambulab.com/v1
[2023-03-18 05:52:53.815188 +01:00] [I] [T 58176]: region update api = https://api.bambulab.com/
[2023-03-18 05:52:53.815188 +01:00] [I] [T 58176]: region update mqtt = ssl://us.mqtt.bambulab.com:8883
[2023-03-18 05:52:53.815188 +01:00] [I] [T 58176]: region update tutk = US
[2023-03-18 05:52:53.815189 +01:00] [I] [T 58176]: region update base domain = bambulab.com
[2023-03-18 05:52:53.815189 +01:00] [I] [T 58176]: region update environment = 
[2023-03-18 05:52:53.815189 +01:00] [I] [T 58176]: region update wifi_code = US
[2023-03-18 05:52:53.815190 +01:00] [I] [T 58176]: region update design = https://bambulab.com
[2023-03-18 05:52:53.815197 +01:00] [I] [T 58176]: Agent: load_config
[2023-03-18 05:52:53.815477 +01:00] [I] [T 58176]: load_user_info
[2023-03-18 05:52:53.815508 +01:00] [I] [T 58176]: set_on_server_connected_fn
[2023-03-18 05:52:53.815509 +01:00] [I] [T 58176]: set_on_printer_connected_fn
[2023-03-18 05:52:53.815510 +01:00] [I] [T 58176]: set_get_country_code_fn
[2023-03-18 05:52:53.815510 +01:00] [I] [T 58176]: set_on_local_connect_fn
[2023-03-18 05:52:53.815510 +01:00] [I] [T 58176]: set_on_message_fn
[2023-03-18 05:52:53.815511 +01:00] [I] [T 58176]: set_on_local_message_fn
[2023-03-18 05:52:53.815517 +01:00] [I] [T 58176]: set_on_http_error_fn
[2023-03-18 05:52:53.815524 +01:00] [I] [T 58176]: ssdp: stop()
[2023-03-18 05:52:53.819242 +01:00] [I] [T 58176]: SsdpDiscovery:start_ssdp with card number = 2
[2023-03-18 05:52:53.819336 +01:00] [I] [T 58176]: SsdpDiscovery::start_discover(), create thread 0
[2023-03-18 05:52:53.819401 +01:00] [I] [T 58176]: SsdpDiscovery::start_discover(), create thread 1
[2023-03-18 05:53:15.861818 +01:00] [I] [T 58176]: connect_cloud_mqtt
[2023-03-18 05:53:17.419137 +01:00] [I] [T 59544]: cloud_conn_callback::on_success, Connection(mqtt) OK! cli id= slicer:2250655135:8ac9
[2023-03-18 05:53:17.419152 +01:00] [I] [T 59544]: cloud_conn_callback::on_success, Connection(mqtt) OK! cli id= slicer:2250655135:8ac9
[2023-03-18 05:53:17.419156 +01:00] [I] [T 59544]: cloud_conn_callback::connected!
[2023-03-18 05:53:17.441030 +01:00] [I] [T 58176]: set monitor machine = MACHINE_ID
[2023-03-18 05:53:17.441052 +01:00] [I] [T 58176]: del_subscribe topic=device/MACHINE_ID/report
[2023-03-18 05:53:17.441262 +01:00] [I] [T 58176]: add_subscribe topic=device/MACHINE_ID/report
[2023-03-18 05:53:17.441265 +01:00] [I] [T 58176]: Agent: save_config
[2023-03-18 05:53:17.791270 +01:00] [I] [T 59544]: subscribe topic: device/MACHINE_ID/report success
[2023-03-18 05:53:17.791296 +01:00] [I] [T 59544]: subscribe return code: 0
[2023-03-18 05:53:17.791297 +01:00] [I] [T 59544]: subscribe reason code: 0
[2023-03-18 05:53:17.792074 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"pushing":{"command":"pushall","sequence_id":"20000"}} 
[2023-03-18 05:53:17.792184 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"info":{"command":"get_version","sequence_id":"20001"}} 
[2023-03-18 06:07:44.673354 +01:00] [I] [T 58176]: stop_subscribe, machine = MACHINE_ID, module = app
[2023-03-18 07:41:22.089900 +01:00] [I] [T 58176]: start_subscribe, machine = MACHINE_ID, model = app
[2023-03-18 07:41:22.090502 +01:00] [I] [T 58176]: add_subscribe topic=device/MACHINE_ID/report
[2023-03-18 07:41:22.268452 +01:00] [I] [T 59544]: subscribe topic: device/MACHINE_ID/report success
[2023-03-18 07:41:22.268468 +01:00] [I] [T 59544]: subscribe return code: 0
[2023-03-18 07:41:22.268470 +01:00] [I] [T 59544]: subscribe reason code: 0
[2023-03-18 07:41:22.268519 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"pushing":{"command":"pushall","sequence_id":"20002"}} 
[2023-03-18 07:41:22.268630 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"info":{"command":"get_version","sequence_id":"20003"}} 
[2023-03-18 07:41:22.786644 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"info":{"command":"get_version","sequence_id":"20004"}} 
[2023-03-18 07:41:33.177159 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"info":{"command":"get_version","sequence_id":"20005"}} 
[2023-03-18 07:41:50.602990 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"info":{"command":"get_version","sequence_id":"20006"}} 
[2023-03-18 07:41:50.603440 +01:00] [I] [T 58176]: publish_json topic=device/MACHINE_ID/request, payload={"pushing":{"command":"pushall","sequence_id":"20007"}} 
[... cut ...]
[2023-03-18 07:46:28.669511 +01:00] [I] [T 14608]: publish_json topic=device/MACHINE_ID/request, payload={"info":{"command":"get_version","sequence_id":"20002"}} 
[2023-03-18 07:46:55.675238 +01:00] [I] [T 14608]: publish_json topic=device/MACHINE_ID/request, payload={"pushing":{"command":"start","sequence_id":"20003"}} 
[2023-03-18 07:47:51.658223 +01:00] [I] [T 14608]: ssdp: stop()
[2023-03-18 07:47:51.864651 +01:00] [I] [T 14608]: SsdpDiscovery::stop_discover(), join thread 0
[2023-03-18 07:47:52.067049 +01:00] [I] [T 14608]: SsdpDiscovery::stop_discover(), join thread 1
[2023-03-18 07:47:52.067156 +01:00] [I] [T 14608]: Agent: save_config
`

StoneLiBambulab, 2023-04-06:
Hi @0rsa ,
    Do you still have this problem?  please check your printer network.
upload your full bambu studio log may help me to debug your issue.
exporting bambu studio log, please refs to wiki page https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks!
Stone Li

SaltWei, 2024-01-19:
Hello, @0rsa 
Does this issue still exist under your PC environment?

0rsa, 2024-01-19:
@SaltWei I didn't experiment this since a long time now; Thanks :)

closed 2024-01-19 by 0rsa

SaltWei, 2024-01-19:
> @SaltWei I didn't experiment this since a long time now; Thanks :)

You are welcome! Please let us known if you met any other new issue