bambulab/BambuStudio #1512: [FEATURE REQUEST] Increase more "Recent models" perhaps tied to Settings/Preferences option?

bolsoncerrado, 2023-03-18:
**Is your feature request related to a problem? Please describe.**
I find it handy to  have MORE recent models listed to choose from when I want to start a new print. Specially useful for repetitive models in a Farm environment.

**Describe the solution you'd like**
Increase more "Recent models" perhaps tied to Settings/Preferences option?

**Describe alternatives you've considered**
None

**Additional context**
None

StoneLiBambulab, 2023-03-19:
Hi @bolsoncerrado ,
    it is a const value in the code, Consider that very few users will use this feature, maby we can provide a way to modify the config file to support this.

Thanks!

Stone Li

bolsoncerrado, 2023-03-20:
OK thanks.

psy0rz, 2023-04-25:
check out https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65

can probably be closed now.

closed 2023-04-26 by StoneLiBambulab