bambulab/BambuStudio #1513: Add Custom G-code on layer show the last custom gcode on every layer

Hologos, 2023-03-18:
**Describe the bug**
I added 2 different custom g-codes on 2 different layers. If I check the custom g-code on both layer, the UI show the last custom g-code even though the correct custom g-code is present in generated g-code.

**3mf File for This Bug**
[at-at_plate_1.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/11009512/at-at_plate_1.gcode.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Add custom g-code on one layer.
2. Add custom g-code on different layer.
3. See custom g-code on both layers.
4. Export g-code and check in generated g-code.

**Expected behavior**
Correct custom g-code is shown in the UI.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.2.1
 - Bambu Studio 1.5.0.61

LiZ-Li-BBL, 2023-03-23:
It's a flaw. Actually you cannot check the added custom g-code. We will add viewing function later. Appreciate your feedback.

LiZ-Li-BBL, 2023-04-21:
In the new release v1.6.0 Public Beta. You can edit the added custom gcode in the layer, or when adding a new custom gcode the inputbox will be empty by default. And also, a gcode-window has been added so that you can view and check the actual inserted gcode.

closed 2023-04-21 by LiZ-Li-BBL