bambulab/BambuStudio #1515: Keyboard shortcuts for Prepare, Preview, Device, Project tabs

Stooovie, 2023-03-19:
**Is your feature request related to a problem? Please describe.**
Please add keyboard shortcuts for the main UI tabs (Prepare, Preview, Device, Project tabs) as well as the Slice command. Using mouse to navigate is is tiresome and slow.

**Describe the solution you'd like**
PrusaSlicer has numbers (1,2,3...) for views and ctrl (or cmd)-numbers (cmd-1, cmd-2...) for UI panels. I like that solution.

KD7LRJ, 2023-03-20:
Ctrl+Tab and Shift+Ctrl+Tab rotate through the tabs and I like this even better than PrusaSlicer most of the time. Ctrl+5, Ctrl+6 are a longer stretch for my left hand most of the time.

Stooovie, 2023-03-20:
thanks, ctrl-tab sort of works but VERY badly on Mac, it sort of goes around one time and then stops and doesn't work anymore in any direction. Probably just buggy. Thanks! (OT: I feel cmd-number on Mac are more user friendly than the Windows ctrl-number analogue as you push cmd with your thumb and reach easier and farther)

MadMax389, 2023-03-20:
Can't use 1-9 as these assign slots to the models. Cntrl 1-6 (Windows) already does slicer view orientation, it just needs to center on the currently active plate. What I REALLY miss from Prusa Slicer are the PageUP/PageDOWN commands to rotate the object +/- 45 degrees on the plate.

danijel1124, 2023-06-02:
Hi,

I would also appreciate key combinations that are, I think it's called deterministic, so it's clear which tab is selected.
ctrl+tab goes through all of them, and that's great if you have to go through all the tabs anyway, but if you have to press that key combination up to four times when you want to go to a specific tab, you probably could have used the mouse.
Additionally, as a blind user, it's much easier for me to have one button that does exactly one thing, since checking which tab is currently in focus is completely eliminated.

KaizenGrit, 2024-11-27:
please add these shortcuts

JonLaliberte, 2025-01-23:
Another request for this. Ctrl+tab doesn't work at all on Mac.

NikolaiRuhe, 2025-02-18:
Please add commands to menu, together with shortcuts.