bambulab/BambuStudio #1516: Imported STEP curves are not always converted to smooth curves in Bambu Studio (lack of precision)

DzzD, 2023-03-19:
**First, being able to use STEP files is great !! but...**

When importing a STEP file, curves are converted to chained faces. 

Sometime this conversion lack of precision (not enought faces), this result in face visible in the Bambu Lab studio but also on the printed part depending on the "Quality=>Precision=>Resolution setting" wich convert back chained faces to arcs when slicing.

A workaround is to increase "Quality=>Precision=>Resolution setting" a bit, but it would be better to improve STEP usage.

**Two solutions :**
1 - dont convert STEP curves to chained faces and use STEP curves for slicing (probably the hardest to implement).
2 - or when converting STEP in chained faces curves use more faces.

Below, a picture of an imported STEP file with two curves, the small one is ok, the big one show faces that will also appear in printed object :

CAD softwar (Fusion360), smooth curves : 
[image]

Bambu Studio, small curve ok, big curve show faces : 
[image]

Printed object, small curve ok, big curve show faces :
[image]

**Finally :** With Fusion 360 exporting STL in high quality give better result than exporting STEP file which is too bad :(

zhimin-zeng-bambulab, 2023-04-11:
@DzzD 
Import step file has a precision control, if the precision is too small, large file import will be very slow.
The current precision can take into account the display effect and import speed. And it does not affect the print quality.

zhimin-zeng-bambulab, 2023-04-11:
@DzzD 
Could you share your step file? I would like to use your file to test the print quality of the model again.

zhimin-zeng-bambulab, 2023-04-14:
@DzzD  I'm sorry that due to the software we are unable to implement it as you want.
Because we use OCCT to process the mesh division that only provides linear deflection and angular deflection to control the division accuracy instead of absolute size.

DzzD, 2023-04-14:
Well, dont be sorry ;) I understand that this would represent a lot of works as you are making the conversion in 3D, because eveything in the studio is made to work with 3D mesh composed of faces, wich imply a lot of restrictions.

It is like if you have made a bitmap image software and that I was asking you to work with vector images, not really feasible without a full redesign/re :)

The best of the best would have been to use the nurbs when slicing to get an exact interprétation of the STEP file curves but that would imply a full rework of the Studio, maybe in the futur you will have more times to handle this and will become also  a leader in slicer ! like with your printer.

Anyway thanks for your time, keep up the good work !

zhimin-zeng-bambulab, 2023-04-17:
Thank you very much for your understanding!

fiing, 2023-06-05:
I don't know if it is 100% the same issue but I have problems too with round edges when imported from .step file.
The item getting some "triangular" edges (1-2 mm in length)
When I import the .step file in other programs the edges is as they should be without these triangular form.

[image]
[image]

[TestGitHub.zip](https://github.com/bambulab/BambuStudio/files/11657610/TestGitHub.zip)

zhimin-zeng-bambulab, 2023-06-06:
@fiing  Thanks! It is a bug for step file import, we will fix it.

zhimin-zeng-bambulab, 2023-06-09:
related to #825

zhimin-zeng-bambulab, 2024-10-21:
[image]

Now, in the process of importing Step files, we have added a precision control window, and users can adjust the import precision according to their own model import effect.

closed 2024-10-21 by zhimin-zeng-bambulab

DzzD, 2024-10-21:
> ![image](https://private-user-images.githubusercontent.com/120697589/378345776-1b36d83d-c9e4-4f1e-afe1-3492864a3391.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

That's perfect ! thanks a lot